projectacrn/acrn-hypervisor #6224: "Launch Windows as the Guest VM on ACRN" tutorial needs to be updated for Tiger Lake (TGL)

gvancuts, 2021-06-21:
The [Launch Windows as the Guest VM on ACRN](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html) tutorial needs to be updated to use the Tiger Lake (TGL) NUC system used in the [Getting Started Guide](https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html).

There is a direct link to this tutorial from the GSG document so this is important.

A couple of important updates that need to be made in the process:
* Reference to a non-existant `acrngt.conf` file (in the GSG)
* Reference to a non-existant `launch_uos_id1.sh` script (in the GSG)

NanlinXie, 2021-06-21:
@gvancuts , I could check with our Dev for the WaaG creation status, but I am afraid that we don't have enough time to refresh this document in v2.5. Two option: 1 is we keep it as known issue. 2 is we delete the launch WaaG in GSG. What's your opinion?

gvancuts, 2021-06-21:
Keep it as a known issue. And add a note in the GSG to that effect so the user is warned.

fuzhongl, 2021-06-28:
Following issue is fixed in [GSG:](https://projectacrn.github.io/latest/getting-started/getting-started.html)

1. Reference to a non-existant acrngt.conf file (in the GSG)

2. Reference to a non-existant launch_uos_id1.sh script (in the GSG)

closed 2021-06-28 by fuzhongl