projectacrn/acrn-hypervisor #6237: PSE port not working in SOS

clefrks, 2021-06-22:
**Describe the bug**
PSE TSN GbEs is not working 
([ 19.756677] stmmaceth 0000:00:1d.1 enp0s29f1: stmmac_request_irq: alloc mac MSI -22 (error: -22)

**Platform**
EHL-CRB

**Codebase**
ACRN version 2.3

**Scenario**
Industry

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://projectacrn.github.io/2.3/getting-started/rt_industry_ubuntu.html
2. Completed ACRN installation
3. BOOT to acrn
4. dmesg | grep eth
5. ([ 19.756677] stmmaceth 0000:00:1d.1 enp0s29f1: stmmac_request_irq: alloc mac MSI -22 (error: -22)

**Expected behavior**
ethernet has IP and can work

**Additional context**
Attach the following logs if available:

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

gvancuts, 2021-06-23:
Hi @clefrks , can you attach the logs you mention?

@NanlinXie , I don't have such platform available to me. Can we get someone from the engineering team to look at this once we have the logs?

clefrks, 2021-06-24:
Attaching the dmesg log
[dmesg-ehl.txt](https://github.com/projectacrn/acrn-hypervisor/files/6706033/dmesg-ehl.txt)

fuzhongl, 2021-06-28:
@clefrks EHL-CRB isn't  ACRN open source platform; please contact with INTEL FAE who supports your project.

clefrks, 2021-06-29:
@fuzhongl EHL-CRB and ehl-crb-b is a different platform ? 
`acrn-hypervisor/misc/config_tools/data/ehl-crb-b/`

fuzhongl, 2021-06-29:
You can generate the board.xml with your EHL platform and compare with the one in acrn-hypervisor/misc/config_tools/data/ehl-crb-b/

BTW: ehl-crb-b is also not ACRN open source platform. please check following link for open source platform:
https://projectacrn.github.io/latest/reference/hardware.html#tested-platforms-by-acrn-release

clefrks, 2021-07-05:
Closing since ehl crb is not open source platform.

closed 2021-07-05 by clefrks