projectacrn/acrn-hypervisor #6238: [config-tools] PTCT/RTCT file need to be well maintained by board_inspector/UI and build script

Kunhui-Li, 2021-06-22:
mail from Matthias:

Subject: ACRN 2.5 board inspector and PTCT/RTCT file

Hello ACRN team,

as discussed in our ACRN sync meeting today, I have now executed the ACRN 2.5 board inspector on my TGL board. As described in the ACRN 2.5 documentation, the board_inspector only generates the $(BOARD).xml file. This is different to the ACRN 2.4 board_parser, who has placed the $(BOARD).xml and the PTCT file in the out directory.

Now I have the same question as raised by Vinay this morning: for ACRN 2.5, is it sufficient to capture only the $(BOARD).xml file or do we also need the PTCT (or RTCT) file, which is available on the target at /sys/firmware/acpi/tables/RTCT ???

The folder https://github.com/projectacrn/acrn-hypervisor/tree/release_2.5/misc/config_tools/acpi_template/tgl-rvp contains a PTCT file. Is it a leftover from ACRN 2.4? Or do we need it also for ACRN 2.5? In the latter case, shall we capture it from the target and rename it from RTCT into PTCT?

Kind regards

Matthias

Kunhui-Li, 2021-06-22:
[External_System_ID] ACRN-7248

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie