projectacrn/acrn-hypervisor #6239: Need documentation for config tools

dbkinder, 2021-06-22:
In ``misc/config_tools/`` there are many folders with configuration tools but no documentation.  There are a few README files but nothing that is prepared to include in our project documentation.

Decide which of these tools need to be documented (e.g., command, purpose, parameters, output, etc.) , create appropriate ``.rst`` content, and integrate into the ACRN documentation.  Alternatively, we need to add more documentation included in or linked to https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html. :
```
acpi_gen
acpi_template
board_config
board_inspector
config_app
data
hv_config
kconfig
launch_config
library
README
scenario_config
schema
static_allocators
xforms
```
There's also the ``kconfig`` folder that's likely obsolete.

gvancuts, 2021-06-23:
> In `misc/config_tools/` there are many folders with configuration tools but no documentation. There are a few README files but nothing that is prepared to include in our project documentation.

The content of some of those is outdated and inaccurate, e.g.:
* `misc/config_tools/README`
```
Kconfig		: Select working scenario and target board, configure ACRN hypervisor capabilities and features.
```
It's `kconfig` (not capital `K`)
```
	target		: Get target board information under native Linux environment and generate board_info XML.
```
That does not exist anymore
```
	board_config	: Parse board_info XML and scenario XML to generate board related configuration files under misc/acrn-config/xmls/board-xmls/ folder.
	scenario_config	: Parse board_info XML and scenario XML to generate scenario based VM configuration files under misc/acrn-config/xmls/config-xmls/$(BOARD)/ folder.
	launch_config	: Parse board_info XML, scenario XML and devicemodel param XML to generate launch script for post-launched vm under misc/acrn-config/xmls/config-xmls/$(BOARD)/ folder.
```
Location needs to be corrected
```
	library		: The folder stores shared software modules or libs for acrn-config offline tool.
```
Incorrect description

* `misc/config_tools/board_inspector/README`
```
board_parser.py will collect all board related info and then generate a board info file for acrn-config host tool usage.
```
Name of the tool needs to be updated (`board_parser.py`) and `acrn-config` needs to be replaced
```
Please run this script under native Linux environment with root privilege.
```
This will become obsolete soon

NanlinXie, 2021-09-19:
@dbkinder , could we close this issue？

dbkinder, 2021-09-20:
@NanlinXie @gvancuts Not until we complete PR #6445

dbkinder, 2022-02-27:
There are still a few README files around, but we document the tools that are exposed to the developers in other documentation.  (Most of these python scripts are called by other python scripts or by a Makefile and aren't intended to be called directly by users.)

closed 2022-02-27 by dbkinder