projectacrn/acrn-hypervisor #6266: [ADL][config_tools]cpu offline method is wrong in rtvm launch script

Kunhui-Li, 2021-06-30:
Environment
NA

HW/Board
ADL-S

Build link
NA

Image info
acrn release_2.5 branch

Steps
1. use config tool to load default adl-s-crb.xml, industry.xml industry_launch_2uos.xml
2. generate launch script.

Expected result
launch uos script generated correctly.

Actual result
the cpu offline method is wrong in rtvm launch script, it will not offline cpu 12/13/14/15.

offline pinned vCPUs from SOS before launch UOS
for i in `ls -d /sys/devices/system/cpu/cpu[12..13..14..15]`; do
online=`cat $i/online`
idx=`echo $i | tr -cd "[1-99]"`
echo cpu$idx online=$online
if [ "$online" = "1" ]; then
echo 0 > $i/online
online=`cat $i/online`
during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
while [ "$online" = "1" ]; do
sleep 1
echo 0 > $i/online
online=`cat $i/online`
done
echo $idx > ${offline_path}/offline_cpu
fi
done
Reproduce rate
100%
# percentage is not acceptable, please use the actually records like 1/20, 2/20 ,etc.

Kunhui-Li, 2021-06-30:
[External_System_ID]ACRN-7257

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie