projectacrn/acrn-hypervisor #6275: [PCI] ACRN sets Interrupt Line config space register to 0 and ReadOnly.

NanlinXie, 2021-07-01:
Environment

ACRN 2.4 Partition Mode / Zephyr guest and MSI (not MSI-X) capable physical device passed through to the OS.

HW/Board

EHL

Steps

- Setup ACRN in partition mode and launch Zephyr as guest. PT a physical PCI device to the guest and assign an IRQ to it by using pcie_msi_vectors_allocate(), pcie_msi_vector_connect(), pcie_msi_enable() Zephyr API.

Expected result

Interrupt line register can be used by above mentioned functions to allocate a free IRQ line to the device and store the information later useage in sub-sequential called functions.

Actual result

Information which interrupt line is allocated to the device is lost as it cannot be stored to the interrupt line register. A later called isr vector connect function or irq enable function relys on a valid register value and maps or enables always a wrong isr/vector/interrupt line to the device.

Reproduce rate

Always

NanlinXie, 2021-07-01:
[External_System_ID]ACRN-7160

yaomichael, 2021-07-01:
ACRN's argument:

```
According to PCI Spec:

7.5.1.1.12 Interrupt Line Register (Offset 3Ch)
The Interrupt Line register communicates interrupt line routing information. The register is read/write and must be implemented by any Function that uses an interrupt pin (see following description). Values in this register are programmed by system software and are system architecture specific. The Function itself does not use this value; rather the value in this register is used by device drivers and operating systems.
7.5.1.1.13 Interrupt Pin Register (Offset 3Dh)
The Interrupt Pin register is a read-only register that identifies the legacy interrupt Message(s) the Function uses (see Section 6.1 for further details). Valid values are 01h, 02h, 03h, and 04h that map to legacy interrupt Messages for INTA, INTB, INTC, and INTD respectively. A value of 00h indicates that the Function uses no legacy interrupt Message(s). The values 05h through FFh are Reserved.

So INTx is an optional feature. Expect Zephyr to fix it by not assuming it's RW
```

nashif, 2021-07-07:
Can you please file a bug in Zephyr GH issues? https://github.com/zephyrproject-rtos/zephyr/issues/new?assignees=&labels=bug&template=bug_report.md&title=

NanlinXie, 2021-07-07:
Created https://github.com/zephyrproject-rtos/zephyr/issues/36765

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie