projectacrn/acrn-hypervisor #6290: GPU Support for SR-IOV based sharing

NanlinXie, 2021-07-12:
The GPU will implement shareable HW interfaces via the PCI SIG Single Root IO Virtualization (SR-IOV) standard.  The GPU shall provide 1 (legacy compatible) Physical Function and 7 assignable Virtual Functions.
A single image graphics kernel mode driver will be capable of running in legacy, physical function and virtual function modes depending on the configuration of the HW and the interface exposed to the domain in which it is running.
GuC firmware and I915 driver shall provide for configuration of relative performance allocations for the PF and each VF
GVT-g shall provide for assignment of individual display PIPEs and individual display PLANEs to particular VF interfaces for direct control.
GuC firmware and I915 driver  shall provide for the static allocation of  GGTT address space (but not aperture) to each VF.
GuC firmware and I915 driver shall provide for performance and event (i.e. QoS violations/resets) on a per function (PF and VF) basis.
GuC firmware and I915 driver shall provide an interface through which system SW can control execution priority of individual functions
GuC firmware and I915 driver shall provide for relative priorities across VF and PF interfaces.  Higher priority P or VF workloads may preempt workloads from lower priority P or VFs.
GuC firmware and I915 driver  shall provide an interface through which system SW can suspend execution of workloads from a given VF

NanlinXie, 2021-07-12:
[External_System_ID] ACRN-5968

NanlinXie, 2021-07-12:
Doc Impact Analysis: 

- [x] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [x] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [x] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

NanlinXie, 2021-07-12:
Need new feature tutorials

NanlinXie, 2021-08-16:
Move it to future update when ready.

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

NanlinXie, 2022-03-31:
Reopen due to document open.

gvancuts, 2022-07-05:
Where is the documentation for this?

fuzhongl, 2022-07-05:
@gvancuts  Sorry for not aware document open.

gvancuts, 2022-07-05:
> @gvancuts Sorry for not aware document open.

This was literally the last comment in this thread... and the "doc: needed" label is set. We cannot close issues that have that label set. This is part of the agreed process.