projectacrn/acrn-hypervisor #6292: [Task]Configuration tool and data refine and clean up.

NanlinXie, 2021-07-12:
Clear up configtool obsolete kconfig and board.xml files
1. Remove Kconfig code
2. Clean up scenario python script, remove the obsolete files that already been replaced by xsl.
3. Add revision for board.xml and scenario.xml files(schema file)

- Define the notation of revision.  1->2 .... 

- Apply to all board.xml and scenario.xml files: attribute: "schema_version"

- Define the rule to update the revision: board.xml , scenario.xml files(schema file)

4. Clean up sample launch script.
5. Change the scenario name as shared/partitioned/hybrid.

NanlinXie, 2021-07-12:
[External_System_ID] ACRN-7263

NanlinXie, 2021-07-12:
Doc Impact Analysis: 

- [x] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [x] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [x] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

dbkinder, 2021-07-12:
@nanlin @junjiemao1 Do need a schema version number field stored with the XML files.   @junjiemao1 was having issues figuring out how to properly patch an "old" XML file into a "new" one. Perhaps we need a proper tool that can scan and bring any XML file to the current expected format and content.  (For example, an older XML file might not have settings for options that were added in a newer version.)

NanlinXie, 2021-11-03:
Configuration tool document updated.

NanlinXie, 2021-11-22:
https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html
https://projectacrn.github.io/latest/tutorials/board_inspector_tool.html
https://projectacrn.github.io/latest/tutorials/acrn_configurator_tool.html

amyreye, 2022-02-17:
@NanlinXie @dbkinder Can we close this issue?

closed 2022-03-04 by fuzhongl