projectacrn/acrn-hypervisor #6295: [v2.5]ACRN Over QEMU/KVM, boot hypervisor failed

ChangZihao, 2021-07-14:
**Describe the bug**
follow "Enable ACRN Over QEMU/KVM", but hypervisor boot failed and return: 
Assertion failed in file arch/x86/vtd.c,line 256 : fatal error

**Platform**
Intel(R) Xeon(R) Silver 4109T CPU @ 2.00GHz , skylake

**Codebase**
tag v2.5

**Scenario**
BOARD=qemu SCENARIO=sdc

**To Reproduce**
follow "Enable ACRN Over QEMU/KVM"

**Additional context**
```
Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Non-UEFI environment detected.
Multiboot Bootloader: GRUB 2.02-2ubuntu8.23
ACRN Hypervisor
[965898773us][cpu=0][(null)][sev=2][seq=1]:HV version 2.5-2021-07-14 03:30:00-2b913c39e DBG (daily tag:acrn-2021w25.5-180000p) sdc@qemu build by ubuntu, start time 965888183us
[965900908us][cpu=0][(null)][sev=2][seq=2]:API version 1.0
[965901620us][cpu=0][(null)][sev=2][seq=3]:Detect processor: Intel(R) Xeon(R) Silver 4109T CPU @ 2.00GHz
ACRN:\>[966488326us][cpu=0][(null)][sev=2][seq=18]:Assertion failed in file arch/x86/vtd.c,line 256 : fatal error
[966498097us][cpu=0][(null)][sev=2][seq=19]:
 delta = (actual_load_address - CONFIG_HV_RAM_START) = 0x0

[966501060us][cpu=0][(null)][sev=2][seq=20]:
Host Stack: CPU_ID = 0

[966503048us][cpu=0][(null)][sev=2][seq=21]:addr(0x1120bae0)	0x000000001120bb00  0x000000000098970e  0x0000000011634040  0x000001c210ba88da

[966506935us][cpu=0][(null)][sev=2][seq=22]:addr(0x1120bb00)	0x000000001120bb40  0x000000001100c9f6  0x0000000011647000  0x0000000011647000

[966509833us][cpu=0][(null)][sev=2][seq=23]:addr(0x1120bb20)	0x0000000011647000  0x0000000011052700  0x0000000000000000  0x000000001163bf00

[966512490us][cpu=0][(null)][sev=2][seq=24]:addr(0x1120bb40)	0x000000001120bbf0  0x0000000011016b70  0x0000000000000000  0x0000000011052700

[966515169us][cpu=0][(null)][sev=2][seq=25]:addr(0x1120bb60)	0x0000000011647000  0x000000000000000f  0x000000001120bc00  0x0000000011052700

[966517842us][cpu=0][(null)][sev=2][seq=26]:addr(0x1120bb80)	0x0000000012803560  0x0000000000000000  0x000000017d580000  0x0000000000000000

[966520514us][cpu=0][(null)][sev=2][seq=27]:addr(0x1120bba0)	0x000000001120bc50  0x03f8000011039027  0x0000000000000008  0x6a9fa975d9aa33e7

[966523172us][cpu=0][(null)][sev=2][seq=28]:addr(0x1120bbc0)	0x0000000000000000  0x0000000000000000  0x0000000011052700  0x0000000000000000

[966525820us][cpu=0][(null)][sev=2][seq=29]:addr(0x1120bbe0)	0x0000000000000000  0x0000000000000000  0x000000001120bc30  0x000000001101777c

[966528462us][cpu=0][(null)][sev=2][seq=30]:addr(0x1120bc00)	0x0000000000000000  0x6a9fa975d9aa33e7  0x000000001120bc50  0x0000000000000000

[966531069us][cpu=0][(null)][sev=2][seq=31]:addr(0x1120bc20)	0x0000000000000000  0x0000000000000000  0x000000001120bc50  0x0000000011017948

[966533721us][cpu=0][(null)][sev=2][seq=32]:addr(0x1120bc40)	0x0000000000000000  0x0000000000000000  0x000000001120bc70  0x00000000110081e8

[966536352us][cpu=0][(null)][sev=2][seq=33]:addr(0x1120bc60)	0x0000000011001256  0x0000000000000000  0x00000000110562b0  0x00000000696e746c

[966539003us][cpu=0][(null)][sev=2][seq=34]:addr(0x1120bc80)	0x0000000000000000  0x0000000000000000  0x0000000100000000  0x0000000000000000

[966541655us][cpu=0][(null)][sev=2][seq=35]:addr(0x1120bca0)	0x0000000000000000  0x0000000000000000  0x000000001120bcb0  0x000000001120bcb0

[966544351us][cpu=0][(null)][sev=2][seq=36]:addr(0x1120bcc0)	0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[966549032us][cpu=0][(null)][sev=2][seq=37]:

[966549962us][cpu=0][(null)][sev=2][seq=38]:Host Call Trace:

[966551192us][cpu=0][(null)][sev=2][seq=39]:----> 0x000000001100c9f6

[966552573us][cpu=0][(null)][sev=2][seq=40]:----> 0x0000000011016b70

[966553933us][cpu=0][(null)][sev=2][seq=41]:----> 0x000000001101777c

[966555318us][cpu=0][(null)][sev=2][seq=42]:----> 0x0000000011017948

[966556688us][cpu=0][(null)][sev=2][seq=43]:----> 0x00000000110081e8

[966558027us][cpu=0][(null)][sev=2][seq=44]:

ACRN:\>
ACRN:\>vm_list

VM_UUID                          VM_ID VM_NAME                          VM_STATE
================================ ===== ================================ ========
ACRN:\>
```

Thanks

gvancuts, 2021-07-15:
I see this problem on 2.4 and a Skylake processor too. I do not know the root cause, nor do I know the workaround but in my case it does not happen *all* the time, which means that I usually reboot the QEMU/KVM VM until I get to a successful boot. It usually happens in less than 5 reboot.

ChangZihao, 2021-07-16:
after rebooting several times，the Hypervisor is successfully booted with some warning(VMX ctrl 0x482/0x48b not fully enabled), and the kernel of the SOS is blocked during initialization.

```
Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Non-UEFI environment detected.
Multiboot Bootloader: GRUB 2.02-2ubuntu8.23
ACRN Hypervisor
[36899919us][cpu=0][(null)][sev=2][seq=1]:HV version 2.5-2021-07-14 03:30:00-2b913c39e DBG (daily tag:acrn-2021w25.5-180000p) sdc@qemu build by ubuntu, start time 36889283us
[36901662us][cpu=0][(null)][sev=2][seq=2]:API version 1.0
[36902255us][cpu=0][(null)][sev=2][seq=3]:Detect processor: Intel(R) Xeon(R) Gold 6148 CPU @ 2.40GHz
ACRN:\>[37375576us][cpu=0][(null)][sev=2][seq=28]:Start VM id: 0 name: ACRN SOS VM
[37376419us][cpu=0][vm0:vcpu0][sev=3][seq=29]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[37377636us][cpu=0][vm0:vcpu0][sev=3][seq=30]:VMX ctrl 0x48b not fully enabled: request 0x40004ab but get0xab

ACRN:\>
ACRN:\>vm_list

VM_UUID                          VM_ID VM_NAME                          VM_STATE
================================ ===== ================================ ========
dbbbd4347a574216a12c2201f1ab0240   0   ACRN SOS VM                      Running
ACRN:\>vm_console 0

----- Entering VM 0 Shell -----
[    0.000000] Linux version 5.4.106-PKT-200203T060100Z-dirty (ubuntu@ubuntu) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #3 SMP PREEMPT Wed Jul 14 19:58:17 PDT 2021
[    0.000000] Command line: console=ttyS0
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x020: 'AVX-512 opmask'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x040: 'AVX-512 Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x080: 'AVX-512 ZMM_Hi256'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[5]:  832, xstate_sizes[5]:   64
[    0.000000] x86/fpu: xstate_offset[6]:  896, xstate_sizes[6]:  512
[    0.000000] x86/fpu: xstate_offset[7]: 1408, xstate_sizes[7]: 1024
[    0.000000] x86/fpu: xstate_offset[9]: 2432, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x2e7, context size is 2440 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000008efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000008f000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000010ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000018800000-0x000000007ffd7fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffd8000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b0000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017d57ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000017d580000-0x000000017fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 1.13.0-1ubuntu1.1 04/01/2014
[    0.000000] Hypervisor detected: ACRN
[    0.000000] last_pfn = 0x17d580 max_arch_pfn = 0x400000000
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.000000] CPU MTRRs all blank - virtualized system.
[    0.000000] last_pfn = 0x7ffd8 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000f5c40-0x000f5c4f]
[    0.000000] check: Scanning 1 areas for low memory corruption
[    0.000000] Using GB pages for direct mapping
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F59A0 000014 (v00 BOCHS )
[    0.000000] ACPI: RSDT 0x000000007FFE2220 000038 (v01 BOCHS  BXPCRSDT 00000001 BXPC 00000001)
[    0.000000] ACPI: FACP 0x000000007FFE1FE0 0000F4 (v03 BOCHS  BXPCFACP 00000001 BXPC 00000001)
[    0.000000] ACPI: DSDT 0x000000007FFE0040 001FA0 (v01 BOCHS  BXPCDSDT 00000001 BXPC 00000001)
[    0.000000] ACPI: FACS 0x000000007FFE0000 000040
[... 31 lines cut ...]
[    0.000000] percpu: Embedded 51 pages/cpu s170280 r8192 d30424 u524288
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 991067
[    0.000000] Kernel command line: console=ttyS0
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3849772K/4027288K available (18436K kernel code, 1954K rwdata, 6888K rodata, 1656Kinit, 6452K bss, 177516K reserved, 0K cma-reserved)
[    0.000000] kmemleak: Kernel memory leak detector disabled
[    0.000000] ftrace: allocating 52557 entries in 206 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[... cut ...]
[    0.000000] random: get_random_bytes called from start_kernel+0x32b/0x4fc with crng_init=0
[    0.000000] Console: colour dummy device 80x25
[    0.000000] printk: console [ttyS0] enabled
[    0.000000] ACPI: Core revision 20190816
[    0.000000] clocksource: hpet: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604467 ns
[    0.000000] APIC: Switch to symmetric I/O mode setup
[    0.000000] x2apic enabled
[    0.000000] Switched APIC routing to physical x2apic.
[    0.000000] ..TIMER: vector=0x30 apic1=0 pin1=2 apic2=-1 pin2=-1
```

fuzhongl, 2021-10-08:
Fixed in v2.6.

closed 2021-10-08 by fuzhongl