projectacrn/acrn-hypervisor #6303: [TGL][RTCT]Input/output error happened when list acpi/tables directory after enabled RTCT in pre-rt

Kunhui-Li, 2021-07-16:
*Environment*

 HV branch ： Rlease_2.5

*Steps:*
 # git clone acrn-hypervisor and check out to release_2.5
 # Copy RTCT from board to ./misc/config_tools/acpi_template/tgl-rvp/RTCT
 # enabled SSRAM in  hybrid_rt scenario file and build HV
 # replace image on board and boot pre-rt and SOS
 # list file in acpi/tables in pre-rt (ls /sys/firmware/acpi/tables/)

 

*Expected result*
 list acpi/table normally

*Actual result*

Input/output error happened when list acpi/tables

  

*Note:* not find the issue if disable ssram 

*Debugging info*

 root@intel-corei7-64:/# ls /sys/firmware/acpi/tables/
ls: reading directory '/sys/firmware/acpi/tables/': Input/output error
----

Kunhui-Li, 2021-07-16:
[External_System_ID] ACRN-7273

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie