projectacrn/acrn-hypervisor #6310: [ADL-S-CRB] [RDT] fail to build hypervisor when enable RDT

Kunhui-Li, 2021-07-20:
**Describe the bug**
fail to build hypervisor when enable RDT

**Platform**
ADL-S-CRB

**HV branch**
master

**Scenario**
Industry.xml

**Additional context**
The log is as follows.

`/home/robot/workspace/adl_test_robot/acrn-hypervisor/build/hypervisor/configs/scenarios/industry/vm_configurations.c:7:
/home/robot/workspace/adl_test_robot/acrn-hypervisor/build/hypervisor/configs/scenarios/industry/misc_cfg.h:55:4: error: ‘U’ undeclared here (not in a funct
ion)
   55 |  { U }
      |    ^
/home/robot/workspace/adl_test_robot/acrn-hypervisor/build/hypervisor/configs/scenarios/industry/vm_configurations.c:23:11: note: in expansion of macro ‘VM0
_VCPU_CLOS’
   23 |   .clos = VM0_VCPU_CLOS,
      |           ^~~~~~~~~~~~~
make[2]: *** [Makefile:43: default] Error 1
make[2]: Leaving directory '/home/robot/workspace/adl_test_robot/acrn-hypervisor/misc/hv_prebuild'
make[1]: *** [Makefile:395: pre_build] Error 2
make[1]: Leaving directory '/home/robot/workspace/adl_test_robot/acrn-hypervisor/hypervisor'
make: *** [Makefile:99: hypervisor] Error 2`

Kunhui-Li, 2021-07-20:
[External_System_ID] ACRN-7298

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie