projectacrn/acrn-hypervisor #6323: Pre-launched VM Boot from ELF

NanlinXie, 2021-07-26:
The hypervisor shall support boot pre-launched VM with kernel image of ELF type, the hypervisor shall parse ELF header to load the kernel image and launch it from correct entry point.

This support shall be configurable.

NanlinXie, 2021-07-26:
[External_System_ID] ACRN-7310

NanlinXie, 2021-07-26:
Doc Impact Analysis: 

- [ ] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [x] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

NanlinXie, 2021-07-26:
The "Other documents" means "[Getting Started Guide for ACRN Hybrid Mode](https://projectacrn.github.io/latest/tutorials/using_hybrid_mode_on_nuc.html)"

gvancuts, 2021-07-26:
This section in our documentation should also be updated accordingly: https://github.com/projectacrn/acrn-hypervisor/blob/master/doc/introduction/index.rst#boot-sequence

jsun26intel, 2021-07-26:
@gvancuts Why? The issue is just related to pre-launched VM kernel image format.

gvancuts, 2021-07-26:
> @gvancuts Why? The issue is just related to pre-launched VM kernel image format.

Oh yes, my bad. So there is no change whatsoever to how we boot ACRN using EFI?

jsun26intel, 2021-07-26:
just make sure the packed prelaunched zephyr image format is in elf format.

amyreye, 2022-02-17:
@NanlinXie @dbkinder Can we close this issue?

closed 2022-03-04 by fuzhongl