projectacrn/acrn-hypervisor #6360: ACRN Makefile missing dependencies

gvancuts, 2021-08-05:
Some of the dependencies between items are missing in the `Makefile`s used in ACRN. More specifically, two are missing:

1. `acrn.bin` needs to be built for the `hypervisor-install` target

To reproduce this:
```
mkdir tmp
DESTDIR=$(pwd)/tmp make BOARD=nuc7i7dnb SCENARIO=industry install
```
Leads to this
```
...
cc  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include/lib  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include/lib/crypto  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include/common  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include/debug  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include/public  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include/dm  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include/hw  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/boot/include  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/boot/include/guest  -I/home/gvancuts/work/acrn-hypervisor/hypervisor/include/arch/x86  -I/home/gvancuts/work/acrn-hypervisor/build/hypervisor/include  -I/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/boards  -I/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/scenarios/industry  -I/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/scenarios/industry -I. -c  -include /home/gvancuts/work/acrn-hypervisor/build/hypervisor/include/config.h -Wall -W -ffunction-sections -fdata-sections -fshort-wchar -ffreestanding -fsigned-char -m64 -mno-mmx -mno-sse -mno-sse2 -mno-80387 -mno-fp-ret-in-387 -mno-red-zone -mpopcnt -nostdinc -nostdlib -fno-common -Werror -Wno-array-bounds -O2 -fpie -fstack-protector-strong -DSTACK_PROTECTOR -fcf-protection=none -mindirect-branch=thunk-extern -mindirect-branch-register -DCONFIG_RETPOLINE -DHV_DEBUG -DPROFILING_ON -fno-omit-frame-pointer  -gdwarf-2 arch/x86/init.c -o /home/gvancuts/work/acrn-hypervisor/build/hypervisor/arch/x86/init.o -MMD -MT /home/gvancuts/work/acrn-hypervisor/build/hypervisor/arch/x86/init.o
ar  crs /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/sys_init_mod.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/arch/x86/init.o
/usr/bin/bash scripts/genld.sh bsp/ld/link_ram.ld.in /home/gvancuts/work/acrn-hypervisor/build/hypervisor/link_ram.ld /home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/config.mk
cc -Wl,-Map=/home/gvancuts/work/acrn-hypervisor/build/hypervisor/acrn.map -o /home/gvancuts/work/acrn-hypervisor/build/hypervisor/acrn.out  -Wl,--gc-sections -nostartfiles -nostdlib -Wl,-n,-z,max-page-size=0x1000 -Wl,--no-dynamic-linker -pie -z noreloc-overflow   -T/home/gvancuts/work/acrn-hypervisor/build/hypervisor/link_ram.ld \
	-Wl,--start-group /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/lib_mod.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/boot_mod.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/hw_mod.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/vp_base_mod.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/vp_dm_mod.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/vp_trusty_mod.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/vp_hcall_mod.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/libdebug.a /home/gvancuts/work/acrn-hypervisor/build/hypervisor/modules/sys_init_mod.a -Wl,--end-group
objcopy -S --section-alignment=0x1000 -O elf32-i386 /home/gvancuts/work/acrn-hypervisor/build/hypervisor/acrn.out /home/gvancuts/work/acrn-hypervisor/build/hypervisor/acrn.32.out
install -D /home/gvancuts/work/acrn-hypervisor/build/hypervisor/acrn.32.out /home/gvancuts/work/acrn-hypervisor/tmp/usr/lib64/acrn/acrn.nuc7i7dnb.industry.32.out
install -D /home/gvancuts/work/acrn-hypervisor/build/hypervisor/acrn.bin /home/gvancuts/work/acrn-hypervisor/tmp/usr/lib64/acrn/acrn.nuc7i7dnb.industry.bin
install: cannot stat '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/acrn.bin': No such file or directory
make[1]: *** [Makefile:383: install] Error 1
make[1]: Leaving directory '/home/gvancuts/work/acrn-hypervisor/hypervisor'
make: *** [Makefile:141: hypervisor-install] Error 2
```

2. The `acrn_mngr.h` file needs to be installed in the build folder before compiling the `devicemodel`

To reproduce this, just patch the code using:
```
$ git diff
diff --git a/hypervisor/Makefile b/hypervisor/Makefile
index 85af66b81..53598aa63 100644
--- a/hypervisor/Makefile
+++ b/hypervisor/Makefile
@@ -378,7 +378,7 @@ PRE_BUILD_DIR := ../misc/hv_prebuild
 .PHONY: all
 all: pre_build $(HV_OBJDIR)/$(HV_FILE).32.out $(HV_OBJDIR)/$(HV_FILE).bin
 
-install: $(HV_OBJDIR)/$(HV_FILE).32.out
+install: $(HV_OBJDIR)/$(HV_FILE).32.out $(HV_OBJDIR)/$(HV_FILE).bin
        install -D $(HV_OBJDIR)/$(HV_FILE).32.out $(DESTDIR)$(libdir)/acrn/$(HV_FILE).$(BOARD).$(SCENARIO).32.out
        install -D $(HV_OBJDIR)/$(HV_FILE).bin $(DESTDIR)$(libdir)/acrn/$(HV_FILE).$(BOARD).$(SCENARIO).bin
        @if [ -e "$(HV_OBJDIR)/acpi" ];then \
```

And re-run the same command as above. It will lead to:
```
...
cc -g -O0 -std=gnu11 -D_GNU_SOURCE -DNO_OPENSSL -m64 -Wall -ffunction-sections -Werror -O2 -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fno-strict-aliasing -fno-delete-null-pointer-checks -fwrapv -fpie -Wno-stringop-truncation -Wno-address-of-packed-member -I/home/gvancuts/work/acrn-hypervisor/devicemodel/include -I/home/gvancuts/work/acrn-hypervisor/devicemodel/include/public -I/home/gvancuts/work/acrn-hypervisor/build/devicemodel/include -I/services -fstack-protector-strong -DDM_DEBUG -c hw/mmio/core.c -o /home/gvancuts/work/acrn-hypervisor/build/devicemodel/hw/mmio/core.o -MMD -MT /home/gvancuts/work/acrn-hypervisor/build/devicemodel/hw/mmio/core.o
[ ! -e /home/gvancuts/work/acrn-hypervisor/build/devicemodel/core/monitor.o ] && mkdir -p /home/gvancuts/work/acrn-hypervisor/build/devicemodel/core/; \
cc -g -O0 -std=gnu11 -D_GNU_SOURCE -DNO_OPENSSL -m64 -Wall -ffunction-sections -Werror -O2 -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fno-strict-aliasing -fno-delete-null-pointer-checks -fwrapv -fpie -Wno-stringop-truncation -Wno-address-of-packed-member -I/home/gvancuts/work/acrn-hypervisor/devicemodel/include -I/home/gvancuts/work/acrn-hypervisor/devicemodel/include/public -I/home/gvancuts/work/acrn-hypervisor/build/devicemodel/include -I/services -fstack-protector-strong -DDM_DEBUG -c core/monitor.c -o /home/gvancuts/work/acrn-hypervisor/build/devicemodel/core/monitor.o -MMD -MT /home/gvancuts/work/acrn-hypervisor/build/devicemodel/core/monitor.o
core/monitor.c:23:10: fatal error: acrn_mngr.h: No such file or directory
   23 | #include "acrn_mngr.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:245: /home/gvancuts/work/acrn-hypervisor/build/devicemodel/core/monitor.o] Error 1
make[1]: Leaving directory '/home/gvancuts/work/acrn-hypervisor/devicemodel'
make: *** [Makefile:171: devicemodel-install] Error 2
```

wenlingz, 2021-08-06:
@gvancuts please help add the JIRA ID, such as [External_System_ID] ACRN-****. You can create one JIRA by yourself.

gvancuts, 2021-08-06:
[External_System_ID] ACRN-7323

gvancuts, 2021-09-23:
PR was merged, closing.

closed 2021-09-23 by gvancuts