projectacrn/acrn-hypervisor #6376: [Doc] Need refresh partitioned guide

NanlinXie, 2021-08-11:
Some feedback from user:

https://projectacrn.github.io/latest/tutorials/using_partition_mode_on_nuc.html

1.  [Update Kernel Image and Modules of Pre-Launched VM](https://projectacrn.github.io/latest/tutorials/using_partition_mode_on_nuc.html#id3) 
point 1, master  `kernel_config_uos` is missing, only existing on release branch. 

2. Need install make，gcc，iasl libelf-dev dependancy.
3. Add make `-j $(nproc)` to speed up compilation. 
4. [Update Kernel Image and Modules of Pre-Launched VM](https://projectacrn.github.io/latest/tutorials/using_partition_mode_on_nuc.html#id3)  
point 1，some files are not used after stored in disk. 

5. [Update Kernel Image and Modules of Pre-Launched VM](https://projectacrn.github.io/latest/tutorials/using_partition_mode_on_nuc.html#id3)  
Need refine or add picture.

6. acrn-hypervisor -> `misc/config_tools/data/whl-ipc-i7/logic_partition.xml` -> `vm1` -> `bootargs` 
root need change to `/dev/nvme0n1p3` （nvme），
need add nvme passthrough to vm1 into pci_devs  
`<pci_dev>02:00.0 Non-Volatile memory controller: Silicon Motion, Inc. Device 2263 (rev 03)</pci_dev>`
otherwise failed to boot.

NanlinXie, 2022-04-14:
Remove this tutorial from latest document, put it to backlog for future improvement.

amyreye, 2022-04-19:
@NanlinXie Here is some feedback from Geoffroy, which he provided during the v2.7 release: 

Geoffroy said: The “Getting Started Guide for ACRN Logical Partition Mode” tutorial does not make sense to me. We talk about three different storage devices (SATA, NVMe and USB) and we seem to unnecessarily copy kernels & modules across.