projectacrn/acrn-hypervisor #6388: NULL function pointer access  risk in virtio_console.c

yonghuah, 2021-08-12:
#1  0x0000555555586e5d in virtio_console_notify_tx (vdev=<optimized out>, vq=0x5555557fe118) at hw/pci/virtio/virtio_console.c:394
#2  0x0000555555561d05 in virtio_poll_timer (arg=0x5555557fddf0, nexp=<optimized out>) at hw/pci/virtio/virtio.c:99
#3  0x00005555555ac66f in timer_handler (fd=<optimized out>, t=<optimized out>, arg=0x5555557fde50) at core/timer.c:59
#4  0x00005555555a7556 in mevent_handle (numev=<optimized out>, kev=<optimized out>) at core/mevent.c:203
#5  mevent_dispatch () at core/mevent.c:441
#6  0x000055555555e799 in main (argc=1, argv=<optimized out>) at core/main.c:1104

yonghuah, 2021-08-12:
[External_System_ID] ACRN-7290

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie