projectacrn/acrn-hypervisor #6411: Need help. ubuntu service vm not booting

STashakkori, 2021-08-17:
Hello and thank you for your help. I bought a Tiger Lake laptop for the sole purpose of running ACRN. Have not had success booting following the "Install the Ubuntu Service VM on the NVMe Disk" instructions. I have installed Ubuntu 20 and am using nvme.

I have built ACRN to completion and am at the step where I'm rebooting after setting the grub2 configuration settings and pointing everything to the kernel. The machine gets suck at "loading ACRN...". I have set the root partition uuid and partid properly. Here are the settings I built ACRN with:
BOARD=nuc11tnbi5    SCENARIO=industry

I am using branch 2.5. I would really like to make this work as I am somewhat invested at this point. Is there any help you can provide to get me going? I realize that ACRN is not verified to run Ubuntu 20 yet I have a suspicious that it is in fact a configuration of the BSP to work with the hardware rather than an issue with the OS at the moment. But then again what do I know? Thank you so much for your assistance. 

Best, Sina

gvancuts, 2021-08-18:
Hi @STashakkori , and great to hear you are joining the ACRN community of users!

I concur with your analysis that Ubuntu 20.04 is unlikely to be the issue here at this point, in fact, this is the version I run most of the time (although not in your particular configuration).

The most likely cause of your troubles here is the delta between the pre-defined `nuc11tnbi5` board XML we provide and your system. You may need to run the board inspector on your system to match its HW configuration (see [here](https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html#using-acrn-configuration-toolset) for instructions on how to do that). ACRN is sensitive to HW differences, especially in terms of memory, total number of CPUs, etc.

STashakkori, 2021-08-19:
Thank you my friend Geoffroy in Belgium. When I run 'sudo python3 cli.py nuc11tnbi5' I get the following error:
IndexError: list index out of range on line 213.
Looking at the generated .xml file I see that the script dies right after writing the DHRD_INFO tag.
Looking at legacy/dmar.py line 213, we are not recognizing the PCI bus.device.function. Specifically dying when parsing the function (fun)
Printing Line, Bus, and Device to get some insight shows this:
***LINE:  0000:00:00.0 Host bridge: Intel Corporation 11th Gen Core Processor Host Bridge/DRAM Registers (rev 01)
***BUS:  0
***DEV:  0
Seems that are we dying on recognizing the Host Bridge itself ie domain0, bus0, device 0, function 0
And can't parse the 0 out of the function. Do you know why this is? Can you help me with this? Perhaps I am missing something here.
I have the following dependencie
Seems that are dying on recognizing the Host Bridge itself ies available:
cpuid, rdmsr, lspci, pip and pip3 recognize lxml, dmidecode, and stty.
Thank you Geofrey and anyone else who can help. Much appreciated. -Sina

STashakkori, 2021-08-19:
I also think I should mention that I'm building Acrn on the host so host is also target ie the $(BOARD) == my host. I hope that's ok..

STashakkori, 2021-08-19:
Ok I'm narrowing in on the issue. When I print fragments of the problem line ie line 213 in legacy/dmar.py I get the following:
line.strip()    **0000:00:00.0 Host bridge: Intel Corporation 11th Gen Core Processor Host Bridge/DRAM Registers (rev 01)**
line.strip().split()[0]    **0000:00:00.0**
line.strip().split()[0].split(:)    **['0000', '00', '00.0']**
line.strip().split()[0].split(:)[1]    **00**
int(line.strip().split()[0].split(:)[1].strip(.), 16)    **0**

But as soon as I assign that to fun I get the error:
fun = int(line.strip().split()[0].split(":")[1].split(".")[1].strip(), 16)
**IndexError: list index out of range**

It is the assignment that is the exposing the issue at the moment. Why this is is something I will investigate further. I'm obviously not a python expert but at first glance its possible the assignment is causing the right side operations to happen in a specific order that is breaking it perhaps.

STashakkori, 2021-08-19:
Ok so if I break the operation in legacy/dmar.py apart into the following:
step1 = line.strip()
step2 = step1.split()[0]
step3 = step2.split(":")[1]
step4 = step3.strip(".")
step5 = int(step4, 16)

And then print:
print("step1    ", step1)
print("step2    ", step2)
print("step3    ", step3)
print("step4    ", step4)
print("step5    ", step5)

I get what we want without any errors:
step1     0000:00:00.0 Host bridge: Intel Corporation 11th Gen Core Processor Host Bridge/DRAM Registers (rev 01)
step2     0000:00:00.0
step3     00
step4     00
step5     0

So I will go ahead and change step5 to be fun and run the script and see what happens. Should this be changed in the repo as well? What do you think? Would this help others or is this just me seeing this issue for some reason?

STashakkori, 2021-08-19:
Ok that worked. If other people bump into that problem perhaps the change should be made in the repo? I don't know maybe I'm the only one who had that problem. Now I'm here:
Warning:The Cx data for ACRN relies on acpi_idle driver but it is not found, please add idle=nomwait in kernel cmdline to fall back to acpi_idle driver
Warning:The Px data for ACRN relies on acpi-cpufreq driver but it is not found, please add intel_pstate=disable in kernel cmdline to fall back to acpi-cpufreq driver
nuc11tnbi5.xml has been generated successfully!
Traceback (most recent call last):
  File "/home/acrn/work/acrn-hypervisor/misc/config_tools/board_inspector/cli.py", line 65, in <module>
    main(args.board_name, board_xml, args)
  File "/home/acrn/work/acrn-hypervisor/misc/config_tools/board_inspector/cli.py", line 47, in main
    module.extract(board_etree)
  File "/home/acrn/work/acrn-hypervisor/misc/config_tools/board_inspector/extractors/60-pci.py", line 150, in extract
    enum_devices(bus_node, PCI_ROOT_PATH)
  File "/home/acrn/work/acrn-hypervisor/misc/config_tools/board_inspector/extractors/60-pci.py", line 134, in enum_devices
    device_node = parse_device(bus_node, p)
  File "/home/acrn/work/acrn-hypervisor/misc/config_tools/board_inspector/extractors/60-pci.py", line 53, in parse_device
    cfg = parse_config_space(device_path)
  File "/home/acrn/work/acrn-hypervisor/misc/config_tools/board_inspector/pcieparser/__init__.py", line 40, in parse_config_space
    extcaps = extended_capabilities(data)
  File "/home/acrn/work/acrn-hypervisor/misc/config_tools/board_inspector/pcieparser/extcaps.py", line 85, in extended_capabilities
    caplist = ExtendedCapabilityListRegister.from_buffer_copy(buf, cap_ptr)
ValueError: Buffer size too small (256 instead of at least 260 bytes)

STashakkori, 2021-08-20:
Ok I've tried alot of different things to get this done but seems I am stuck at customization (step 2).

I was able to create the board.xml using App.py but I can't get past this issue. I simply can't find anywhere where I can customize the pCPUs other than the affinity using this tool and I can't set it in board.xml. There must be somewhere else. Please assist:

"The physical CPUs allocated to the same VM shall have the same core types.\n\nOn platforms having both big and little cores, the current design of ACRN only allows allocating the same type of cores\nto a pre-launched VM.\n\nTo fix this error, double check the allocation of physical CPUs to each pre-launched VM and ensure that only big cores\nor little cores are assigned, but not both."

I also have no option to generate a launch setting. I can import and I see that there are some launch xml in the generic_board/ directory but I'm not certain I should use this. I was hoping to create one. Anyway any help would be really awesome. Thank you so much. I think that perhaps I am close to getting this

NanlinXie, 2021-08-20:
Some suggestion:
1. How did you fix the previous issue when generate board.xml? It is strange that seems you have one PCIe device without extended configuration space.  But if you could generate your board.xml successfully, could you send out your board.xml?
2. For the scenario.xml, you could “Load a default scenario” and choose “industry” firstly, just need to take care of the parameter that whether use SATA or NVME.  If you don’t have serial port, it will be difficult for future debug.
3. You could ignore launch setting in this phase, try build and run ACRN firstly.

Best wishes!
Nanlin

gvancuts, 2021-08-20:
> I also think I should mention that I'm building Acrn on the host so host is also target ie the $(BOARD) == my host. I hope that's ok..

Yes, that's perfectly fine

> 
> Also I'm using Python 3 version 3.9.5. I hope thats also ok..

I have not tried with that version. What is your host OS? I'm using Ubuntu 20.04 but that comes with Python 3 version 3.8.10

gvancuts, 2021-08-20:
> Ok that worked. If other people bump into that problem perhaps the change should be made in the repo?
> I don't know maybe I'm the only one who had that problem.

I have never seen that error, and just tried on my Ubuntu 20.04 but it works correctly. I'm wondering what may be triggering this. Do you have any idea? (I'm no Python expert either...).

I think it would be worth getting this in the main repo, although I'd feel more comfortable if we understood first the root-cause of this.

STashakkori, 2021-08-20:
> I have never seen that error, and just tried on my Ubuntu 20.04 but it works correctly. I'm wondering what may be triggering this. Do you have any idea? (I'm no Python expert either...).
>
> I think it [...]

Thank you Geoffroy, I am using Ubuntu 21.04 and Python 3.9.5. I will investigate the issue further this weekend to make sure we understand the exact reason I get it and others don't.

I'm stuck in customization. I'll post in response to @NanlinXie to see if we can get past the current blocker. Thank you and please stand by

STashakkori, 2021-08-20:
1.
@NanlinXie here is a gist of my board.xml:
<script src="https://gist.github.com/STashakkori/82c3187a191a236ff646bdc5ab0301a0.js"></script>

I was able to generate it by doing edits mentioned earlier in this thread, by editing the dmar.py script slightly

2. I will try with the industry scenario choosing NVME. I do not have a serial port unfortunately. I am on a laptop. In fact finding any Tiger Lake right now that is not a laptop is practically impossible.

3. I will take your suggestion and not worry about a launch setting. I will try to build with the default industry scenario and see what happens. I would like to understand how to set big and little pCPUs at some point though as I am interested in becoming an ACRN poweruser as I think this has alot of potential.

Thank you and stand by Nanlin

STashakkori, 2021-08-20:
Ok status now is that I was able to build ACRN using the generated board.xml with default scenario but could not use App.py in any capacity as even with the default scenario, exporting gave me the error documented above: "The physical CPUs allocated to the same CPU...".

Unfortunately, even after building with the custom board.xml I was unable to boot ACRN.

Right now I need to know what success looks like. I will repeat all steps on the target hardware after installing Ubuntu 20.04 since @gvancuts has confirmed he was able to boot this version under ACRN. Once I establish a working baseline I will go back and investigate root cause and how to make this work with my unsuccessful combination of Ubuntu 21.04 and Python 3.9.5.

Stand by and thank you both

STashakkori, 2021-08-21:
Ok @gvancuts and @NanlinXie I've solved our riddle. I have not yet finished writing the solution nor have I tried booting but after seeing the same behavior on Ubuntu 20.04 and Python 3.8.2, I realized something else is going on. And so here it is:

The problem is that this Tiger Lake laptop has two PCI roots ie two PCI domains that will show up when lspci is executed. As you both know, the vast majority of boards have only one PCI root and so this issue did not show up. But on this less common machine, it is a problem because the dmar.py script executes an lspci -xxx and then parses out the token in the second column that is the function and this is a problem because lspci and lspci -xxx will automatically print domains as the first column if it detects more than one root, entailing that the function will reside in the third column instead of the second. So when a split on '.' character and then second element array access is attempted on the second column of my machine's lspci output, it throws the error because that token does not infact exist.

Fun fact it is nvme that resides on my second domain. Which is pretty important from ACRN's point of view and may explain why I can't boot. It is possible that more Intel boards in the future will have this behavior and so it may be a good idea to future proof the script.

Here is what I propose. Instead of adding conditional logic to the parser, I think it would be best to instead of just executing lspci -xxx, how about we execute an lspci -xxxD. The -D will force printing of domains. Then we make small adjustments to the parsing logic to point to one column shifted over and ensure that we can always expect the same input line. Thoughts? I will write a fix on my end tomorrow so I can continue and see if I can boot but if you want we can implement this for everyone. Best, Sina

STashakkori, 2021-08-21:
Sigh. So in dmar.py if I change lspci -xxx to be lspci -xxxD and then increment the indices in dev and fun, then I can build the board.xml.

However, unfortunately, because the lspci output gets written to the board.xml itself and used later during the building process, it causes a cascade of problems down the road and so I can't build ACRN.

I think this is pretty definitive. ACRN does not currently support boards with more than one PCI root. Hopefully this will happen sometime soon.

STashakkori, 2021-08-22:
Yea I think this is a blocker for now, going to take a break and look at possible alternatives. I've linked another gist below so you can visually see. Specifically look at <PCI_VID_PID>. This is after I made the small edits to dmar.py so I could build the xml. This config won't work to build ACRN but at least shows the two PCI domains and likely until ACRN has support for this boards like this will not build or boot the system. In general I think the ACRN build system shows great promise. Thank you both for your help, I learned a bit on this journey.

https://gist.github.com/STashakkori/fa1f9f18ad212b0368bb580e9ae39ce3/edit

NanlinXie, 2022-01-07:
[External_System_ID] ACRN-7430

fuzhongl, 2022-07-05:
@STashakkori You are right, ACRN  supports  only one PCI segment so far.
Please workaround this issue by disable the second PCI segment in BOIS.

closed 2022-07-05 by fuzhongl