projectacrn/acrn-hypervisor #6412: Add the per-VM vCAT support for ACRN

donsheng, 2021-08-17:
vCAT overview:

Make CAT feature visible to VMs

CDP/MBA/CMT are not supported and related capabilities are not exposed to guest, 
only support vCAT for last level cache (LLC).

vCAT is enabled on a per-VM basis

Shared cache partition (physical cache ways and CLOSids) is not supported for vCAT-enabled VMs,
e.g. the physical cache ways/CLOSids used by a vCAT-enabled VM should be dedicated and
should not overlap with the physical cache ways/CLOSids used by any other VMs. Note that
for vCAT, cache partition is per-VM, but not per-cpu.

The vCAT configuration can be done in scenario XML file under the FEATURES/VCAT and vm/vcat
 sub-sections.

donsheng, 2021-08-17:
[External_System_ID] ACRN-6101

NanlinXie, 2021-09-13:
Duplicated with https://github.com/projectacrn/acrn-hypervisor/issues/5917

closed 2022-03-04 by fuzhongl