projectacrn/acrn-hypervisor #6428: [acrn-configuration-tool] Fail to generate launch script when disable CPU sharing

Kunhui-Li, 2021-08-18:
**HV branch:**
release_2.5

**Issue description:**
After applying patch, it will not allow user to generate launch script:
`(venv) root@ssghv:/home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor# ../../generate_script.sh
++ board=tgl-rvp
++ xml_path=misc/config_tools/data/tgl-rvp
++ scenario=industry
++ export board_file=/home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/tgl-rvp.xml
++ board_file=/home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/tgl-rvp.xml
++ export scenario_file=/home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/industry.xml
++ scenario_file=/home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/industry.xml
++ export launch_file=/home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/industry_launch_6uos.xml
++ launch_file=/home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/industry_launch_6uos.xml
++ python /home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/launch_config/launch_cfg_gen.py --board /home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/tgl-rvp.xml --scenario /home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/industry.xml --launch /home/rui/workspace/virtualization.hypervisors.acrn.acrn-test.acrn-robot/acrn-hypervisor/misc/config_tools/data/tgl-rvp/industry_launch_6uos.xml --uosid 0
Error: vm:id=2,pcpu_id: The same pcpu was configurated in <pcpu_id>/<cpu_affinity>, but CPU sharing is disabled by 'SCHED_NOOP'. Please re-configurate them!`

Kunhui-Li, 2021-08-18:
[External_System_ID] ACRN-7251

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie