projectacrn/acrn-hypervisor #6431: virtio_console use-after-free

Ragnarliu, 2021-08-19:
#4 0x55adb6da6885 in vm_init_vdevs core/main.c:495
 #5 0x55adb6daa512 in main core/main.c:1040
 #6 0x7f98f446f09a in __libc_start_main ../csu/libc-start.c:308
SUMMARY: AddressSanitizer: heap-use-after-free hw/pci/virtio/virtio_console.c:1064 in virtio_console_close_all
 Shadow bytes around the buggy address:
 0x0c427fff9080: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff9090: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff90a0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff90b0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff90c0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 =>0x0c427fff90d0:[fd]fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff90e0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff90f0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff9100: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff9110: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 0x0c427fff9120: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
 Shadow byte legend (one shadow byte represents 8 application bytes):
 Addressable: 00
 Partially addressable: 01 02 03 04 05 06 07
 Heap left redzone: fa
 Freed heap region: fd
 Stack left redzone: f1
 Stack mid redzone: f2
 Stack right redzone: f3
 Stack after return: f5
 Stack use after scope: f8
 Global redzone: f9
 Global init order: f6
 Poisoned by user: f7
 Container overflow: fc
 Array cookie: ac
 Intra object redzone: bb
 ASan internal: fe
 Left alloca redzone: ca
 Right alloca redzone: cb
 Shadow gap: cc

Ragnarliu, 2021-08-19:
[External_System_ID] ACRN-7103

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie