projectacrn/acrn-hypervisor #6473: HV can't be used after dumpreg rtvm vcpu

Ragnarliu, 2021-08-24:
HW/Board

adl-s-crb

Build link
EB: shuo’s upstream(https://github.com/shuox/acrn-hypervisor/tree/upstream) + Kernel 5.10(http://sit-image.sh.intel.com/yocto/e2e/5.10/yocto/ww25/)
Image info

SOS kernel:Linux intel-corei7-64 5.10.26-acrn-kernel-sos 

HV:HV 2.6-unstable-2021-07-02 15:48:11-20e3b0d3f DBG (daily tag: ) industry@adl-s-crb build by shuang

DM version is: 2.6-unstable-20e3b0d3f (daily tag, build by shuang@2021-07-02 15:48:21

rtvm kernel:5.4.102-rt53-linux-intel-ese-lts-preempt-rt-acrn-uos-lts-rt

 

Steps

1.launch rtvm

2.input "vcpu_dumpreg 2 1" on HV        (vcpu_dumpreg <vm id, vcpu id>)

Expected result

results printed and HV is normal

Actual result

no results printed and HV can't be used,but sos and rtvm are normal

Reproduce rate

100%

Ragnarliu, 2021-08-24:
[External_System_ID] ACRN-7275

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie