projectacrn/acrn-hypervisor #6491: Can not boot Service OS on a custom board

m8, 2021-08-25:
**Issue**
Hi, I'm following the [GSG](https://projectacrn.github.io/latest/getting-started/getting-started.html), but I'm using custom board instead of NUCs. At the end of the GSG, I stuck in booting SOS. Machine waits at the `loading ACRN..` endlessly.

**Platform**
Manufacturer: Hewlett-Packard
Product Name: 18E4
"Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz"
[my_board.txt](https://github.com/projectacrn/acrn-hypervisor/files/7045397/my_board.txt)

**Codebase**
- ACRN Kernel 2.6
- ACRN Hypervisor branch a7c273996

**Scenario**
Industry

gvancuts, 2021-08-25:
Hi @m8 , this is a rather old processor (as far as ACRN is concerned, the project was launched in 2018 and that processor had already been discontinued by then).

The fact it boots directly to Ubuntu after a reboot may be a feature (or an odd behaviour) from the bios.

Do you have a serial port on this platform? This would greatly help us understand what is going on.

Can you also share your `/etc/grub.d/40_custom` and output of `sudo lsblk` on your platform?

m8, 2021-08-25:
thanks for your reply, this my files:
```
lab@lab-HP-EliteDesk-800-G1-TWR:~$ sudo blkid /dev/sda1 
loop0    7:0    0   2,2M  1 loop /snap/gnome-system-monitor/148
loop1    7:1    0 255,6M  1 loop /snap/gnome-3-34-1804/36
loop2    7:2    0   956K  1 loop /snap/gnome-logs/100
loop3    7:3    0  29,9M  1 loop /snap/snapd/8542
loop4    7:4    0  55,3M  1 loop /snap/core18/1885
loop5    7:5    0   276K  1 loop /snap/gnome-characters/550
loop6    7:6    0  62,1M  1 loop /snap/gtk-common-themes/1506
loop7    7:7    0   2,4M  1 loop /snap/gnome-calculator/748
sda      8:0    0 447,1G  0 disk 
├─sda1   8:1    0   512M  0 part /boot/efi
└─sda2   8:2    0 446,6G  0 part /
sdb      8:16   1  14,3G  0 disk /media/lab/9448-6CF8
sr0     11:0    1  1024M  0 rom 

lab@lab-HP-EliteDesk-800-G1-TWR:~$ sudo blkid /dev/sda2
/dev/sda2: UUID="ee79448a-3ac9-4ae6-a994-158e746f834e" TYPE="ext4" PARTUUID="8fbfc9f3-a315-40a7-81f4-ac5001d46b21"

lab@lab-HP-EliteDesk-800-G1-TWR:~$ cat /etc/grub.d/40_custom 
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.

menuentry "ACRN Multiboot Ubuntu Service VM" --id ubuntu-service-vm {
  load_video
  insmod gzio
  insmod part_gpt
  insmod ext2
  search --no-floppy --fs-uuid --set ee79448a-3ac9-4ae6-a994-158e746f834e
  echo 'loading ACRN...'
  multiboot2 /boot/acrn/acrn.bin  root=PARTUUID="8fbfc9f3-a315-40a7-81f4-ac5001d46b21"
  module2 /boot/vmlinuz-5.10.52-acrn-sos Linux_bzImage
}
```
I connected serial port, but couldn't get any logs to development computer, should I enable something for that? I will look to it.

gvancuts, 2021-08-25:
> thanks for your reply, this my files:

Thanks, it all looks correct to me 👍 

> I connected serial port, but couldn't get any logs to development computer, should I enable something for that? I will look to it.

You appear to have two on your system, based on your `my_board.txt` file:
```
  <TTYS_INFO>
	seri:/dev/ttyS0 type:portio base:0x3F8 irq:4
	seri:/dev/ttyS4 type:portio base:0xF0E0 irq:19
	</TTYS_INFO>
```

You need to make sure that you use the right one in your `industry.xml` file (in the `hv` section), e.g.:
```
<SERIAL_CONSOLE>/dev/ttyS0</SERIAL_CONSOLE>
```

Do you know which one you are connected to? Are you able to see the Bios output on the serial port?

You can also overwrite the settings that were used at build time by adding the right parameters to your Grub config, `multiboot` line. There are some instructions that can help you with that here (you'll have to adapt them to your board/serial port settings): https://projectacrn.github.io/latest/tutorials/using_serial_port.html

m8, 2021-08-25:
I got it, my cable is broken, this is my serial output.

```
[13:26:25:318] UEFI environment detected.␊
[13:26:25:318] ␍Multiboot2 Bootloader: GRUB 2.02-2ubuntu8.17␊
[13:26:25:318] ␍detect_hardware_support, SMAP not supported␊
[13:26:25:318] ␍[0us][cpu=0][(null)][sev=1][seq=1]:PANIC: init_pcpu_pre line: 152␊
[13:26:25:318] ␍␊
[13:26:25:318] ␍␍[0us][cpu=0][(null)][sev=1][seq=2]:hardware not support!␊
```

my industry.xml file,
```xml
<acrn-config board="my_board" scenario="industry">
  <hv>
    <DEBUG_OPTIONS>
      <RELEASE>n</RELEASE>
      <SERIAL_CONSOLE>/dev/ttyS0</SERIAL_CONSOLE>
      <MEM_LOGLEVEL>5</MEM_LOGLEVEL>
      <NPK_LOGLEVEL>5</NPK_LOGLEVEL>
      <CONSOLE_LOGLEVEL>3</CONSOLE_LOGLEVEL>
      <LOG_DESTINATION>7</LOG_DESTINATION>
      <LOG_BUF_SIZE>0x40000</LOG_BUF_SIZE>
    </DEBUG_OPTIONS>
    <FEATURES>
      <RELOC>y</RELOC>
      <SCHEDULER>SCHED_BVT</SCHEDULER>
      <MULTIBOOT2>y</MULTIBOOT2>
      <ENFORCE_TURNOFF_AC>y</ENFORCE_TURNOFF_AC>
      <ENFORCE_TURNOFF_GP>n</ENFORCE_TURNOFF_GP>
      <SECURITY_VM_FIXUP>n</SECURITY_VM_FIXUP>
      <RDT>
        <RDT_ENABLED>n</RDT_ENABLED>
        <CDP_ENABLED>y</CDP_ENABLED>
        <CLOS_MASK>0xfffff</CLOS_MASK>
        <CLOS_MASK>0xfffff</CLOS_MASK>
        <CLOS_MASK>0xfffff</CLOS_MASK>
        <CLOS_MASK>0xfffff</CLOS_MASK>
        <CLOS_MASK>0xfffff</CLOS_MASK>
        <CLOS_MASK>0xfffff</CLOS_MASK>
        <CLOS_MASK>0xfffff</CLOS_MASK>
        <CLOS_MASK>0xfffff</CLOS_MASK>
      </RDT>
      <NVMX_ENABLED>n</NVMX_ENABLED>
      <HYPERV_ENABLED>y</HYPERV_ENABLED>
      <IOMMU_ENFORCE_SNP>n</IOMMU_ENFORCE_SNP>
      <ACPI_PARSE_ENABLED>y</ACPI_PARSE_ENABLED>
      <L1D_VMENTRY_ENABLED>n</L1D_VMENTRY_ENABLED>
      <MCE_ON_PSC_DISABLED>n</MCE_ON_PSC_DISABLED>
      <IVSHMEM>
        <IVSHMEM_ENABLED>n</IVSHMEM_ENABLED>
        <IVSHMEM_REGION></IVSHMEM_REGION>
      </IVSHMEM>
    </FEATURES>
    <MEMORY>
      <STACK_SIZE>0x2000</STACK_SIZE>
      <HV_RAM_SIZE></HV_RAM_SIZE>
      <HV_RAM_START></HV_RAM_START>
      <LOW_RAM_SIZE>0x00010000</LOW_RAM_SIZE>
      <PLATFORM_RAM_SIZE>0x400000000</PLATFORM_RAM_SIZE>
    </MEMORY>
    <CAPACITIES>
      <IOMMU_BUS_NUM>0x100</IOMMU_BUS_NUM>
      <MAX_EFI_MMAP_ENTRIES>256</MAX_EFI_MMAP_ENTRIES>
      <MAX_IR_ENTRIES>256</MAX_IR_ENTRIES>
      <MAX_IOAPIC_NUM>1</MAX_IOAPIC_NUM>
      <MAX_PCI_DEV_NUM>96</MAX_PCI_DEV_NUM>
      <MAX_IOAPIC_LINES>120</MAX_IOAPIC_LINES>
      <MAX_PT_IRQ_ENTRIES>256</MAX_PT_IRQ_ENTRIES>
      <MAX_MSIX_TABLE_NUM></MAX_MSIX_TABLE_NUM>
[... 304 lines cut ...]
    </epc_section>
    <legacy_vuart id="0">
      <type>VUART_LEGACY_PIO</type>
      <base>INVALID_COM_BASE</base>
      <irq>COM1_IRQ</irq>
    </legacy_vuart>
    <legacy_vuart id="1">
      <type>VUART_LEGACY_PIO</type>
      <base>INVALID_COM_BASE</base>
      <irq>COM2_IRQ</irq>
      <target_vm_id>0</target_vm_id>
      <target_uart_id>0</target_uart_id>
    </legacy_vuart>
    <console_vuart id="0">
      <base>INVALID_PCI_BASE</base>
    </console_vuart>
    <communication_vuart id="1">
      <base>INVALID_PCI_BASE</base>
      <target_vm_id>1</target_vm_id>
      <target_uart_id>1</target_uart_id>
    </communication_vuart>
    <PTM>n</PTM>
  </vm>
</acrn-config>
```

m8, 2021-08-25:
what can be cause this error?

gvancuts, 2021-08-25:
> what can be cause this error?

This is the problem:
```
[13:26:25:318] ␍detect_hardware_support, SMAP not supported␊
```

The `detect_hardware_support` function fails here: https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/arch/x86/cpu_caps.c#L503-L506 ... and because of that the hypervisor stops.

I don't if this is something that could be turned off because of a Bios setting (if you want to try, go through all options related to Virtualization and make sure it's all turned on). Otherwise I'm afraid it means ACRN cannot run on that platform.

@junjiemao1 , I think this is a good example of how we could improve our configuration tool by catching this as soon as we have the `${BOARD}.xml` file available. The lack of `smap` can be detected in there.