projectacrn/acrn-hypervisor #6518: ]acrn-dm program crash when probe_disk_log_file() fails.

yonghuah, 2021-08-30:
+_debuglog:_+
(gdb) bt
#0 0x00005555555c4caa in printf_common(void*, char const*, __va_list_tag*) ()
#1 0x00005555555f6946 in vasprintf ()
#2 0x0000555555827ed9 in write_to_kmsg (fmt=<optimized out>, args=0x7ffeae1b2150) at log/kmsg_logger.c:79
#3 0x00005555558279f6 in output_log (level=level@entry=1 '\001', fmt=fmt@entry=0x55555589b300 "disk_log: open %s failed! Error: %s\n") at log/log.c:98
#4 0x00005555558285c3 in probe_disk_log_file () at log/disk_logger.c:76
#5 0x0000555555828f1e in write_to_disk (fmt=<optimized out>, args=0x7ffeae1b2750) at log/disk_logger.c:156
#6 0x00005555558279f6 in output_log (level=level@entry=1 '\001', fmt=fmt@entry=0x55555589b300 "disk_log: open %s failed! Error: %s\n") at log/log.c:98
#7 0x00005555558285c3 in probe_disk_log_file () at log/disk_logger.c:76
#8 0x0000555555828f1e in write_to_disk (fmt=<optimized out>, args=0x7ffeae1b2d50) at log/disk_logger.c:156
#9 0x00005555558279f6 in output_log (level=level@entry=1 '\001', fmt=fmt@entry=0x55555589b300 "disk_log: open %s failed! Error: %s\n") at log/log.c:98
#10 0x00005555558285c3 in probe_disk_log_file () at log/disk_logger.c:76
#11 0x0000555555828f1e in write_to_disk (fmt=<optimized out>, args=0x7ffeae1b3350) at log/disk_logger.c:156
#12 0x00005555558279f6 in output_log (level=level@entry=1 '\001', fmt=fmt@entry=0x55555589b300 "disk_log: open %s failed! Error: %s\n") at log/log.c:98
#13 0x00005555558285c3 in probe_disk_log_file () at log/disk_logger.c:76
#14 0x0000555555828f1e in write_to_disk (fmt=<optimized out>, args=0x7ffeae1b3950) at log/disk_logger.c:156
#15 0x00005555558279f6 in output_log (level=level@entry=1 '\001', fmt=fmt@entry=0x55555589b300 "disk_log: open %s failed! Error: %s\n") at log/log.c:98
#16 0x00005555558285c3 in probe_disk_log_file () at log/disk_logger.c:76
#17 0x0000555555828f1e in write_to_disk (fmt=<optimized out>, args=0x7ffeae1b3f50) at log/disk_logger.c:156
#18 0x00005555558279f6 in output_log (level=level@entry=1 '\001', fmt=fmt@entry=0x55555589b300 "disk_log: open %s failed! Error: %s\n") at log/log.c:98
#19 0x00005555558285c3 in probe_disk_log_file () at log/disk_logger.c:76
#20 0x0000555555828f1e in write_to_disk (fmt=<optimized out>, args=0x7ffeae1b4550) at log/disk_logger.c:156

yonghuah, 2021-08-30:
[External_System_ID] ACRN-7364

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie