projectacrn/acrn-hypervisor #6520: compiling not working - master branch

ionutnechita, 2021-08-30:
Hi ACRN Team,

I try to compile with latest commit but not working.

commit 7b19c8e87803f71f0e94587d0e63b95ff2250b46 (HEAD -> master, origin/master, origin/HEAD)
Author: Kunhui-Li <kunhuix.li@intel.com>
Date:   Thu Aug 26 15:58:29 2021 +0800

    config_tools: update the logic of getting gpu bdf

    The bdf of gpu is not 00:02.0 for the new platform icx-rvp,
    it is 05:00.0 now, so we remove the hardcode 00:02.0;
    And change to get the gpu bdf from board.xml.

    Tracked-On: #6357
    Signed-off-by: Kunhui-Li <kunhuix.li@intel.com>

Error:
```
/acrn-hypervisor/misc/debug_tools/acrn_crashlog/acrnprobe/include -I /usr/include/libxml2 -I /home/demo/git/acrn-hypervisor/build/misc/debug_tools/acrn-crashlog/include/acrnprobe -fdata-sections -fcommon vmrecord.c -o /home/demo/git/acrn-hypervisor/build/misc/debug_tools/acrn-crashlog/acrnprobe/obj/vmrecord.o
cp ./acrnd.service /home/demo/git/acrn-hypervisor/build/misc/services/acrnd.service
loop.c: In function ‘loopdev_check_parname’:
loop.c:234:33: error: ‘strcmp’ argument 1 declared attribute ‘nonstring’ [-Werror=stringop-overread]
  234 |                         return !strcmp((const char *)super.s_volume_name, parname);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/ext2fs/ext2fs.h:85,
                 from loop.c:19:
/usr/include/ext2fs/ext2_fs.h:699:17: note: argument ‘s_volume_name’ declared here
  699 | /*078*/ __u8    s_volume_name[EXT2_LABEL_LEN] __nonstring;      /* volume name, no NUL? */
      |                 ^~~~~~~~~~~~~
make[2]: Leaving directory '/home/demo/git/acrn-hypervisor/misc/services/acrn_manager'
cc1: all warnings being treated as errors
```

Information OS:

```
NAME="openSUSE Tumbleweed"
//# VERSION="20210828"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20210828"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20210828"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo"
uname -a
Linux demo-acrn-opensuse-labl1 5.13.12-2-default #1 SMP Fri Aug 20 07:04:28 UTC 2021 (33df9c6) x86_64 x86_64 x86_64 GNU/Linux
gcc --version
gcc (SUSE Linux) 11.2.1 20210816 [revision 056e324ce46a7924b5cf10f61010cf9dd2ca10e9]
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
ldd --version
ldd (GNU libc) 2.33
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
 iasl -v
Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20210604
Copyright (c) 2000 - 2021 Intel Corporation
```

Full log:
```
[... 78 lines cut ...]
Generating vm_configurations.h:
vm_configurations.h was generated using xsltproc successfully.
Generating pt_intx.c:
make[3]: Leaving directory '/home/demo/git/acrn-hypervisor/misc/debug_tools/acrn_crashlog/acrnprobe'
make[2]: *** [Makefile:93: all] Error 2
make[2]: Leaving directory '/home/demo/git/acrn-hypervisor/misc/debug_tools/acrn_crashlog'
make[1]: *** [Makefile:39: acrn-crashlog] Error 2
make[1]: Leaving directory '/home/demo/git/acrn-hypervisor/misc'
make: *** [Makefile:127: tools] Error 2
make: *** Waiting for unfinished jobs....
pt_intx.c was generated using xsltproc successfully.
Generating ivshmem_cfg.h:
ivshmem_cfg.h was generated using xsltproc successfully.
Generating misc_cfg.h:
misc_cfg.h was generated using xsltproc successfully.
Generating pci_dev.c:
pci_dev.c was generated using xsltproc successfully.
Generating board_info.h:
board_info.h was generated using xsltproc successfully.
/usr/bin/clang-format
make[1]: Leaving directory '/home/demo/git/acrn-hypervisor/hypervisor'

```

A nice day.

gvancuts, 2021-08-30:
Looks like it's the same issue as #6494

gvancuts, 2021-09-03:
Hi @ionutnechita , can you take a look at #6535 and confirm if this solves your issue?

ionutnechita, 2021-09-14:
Hi @gvancuts,

I tested now with all merged modification and this resolve my issue.
Thanks for support.

A good day.
I will close this ticket.

closed 2021-09-14 by ionutnechita