projectacrn/acrn-hypervisor #6631: [KATA][5.10 Kernel]failed to start docker with  ServiceVM 5.10  kernel

fuzhongl, 2021-09-28:
**Describe the bug**
A clear and concise description of what the bug is.

**Platform**
WHL

**Codebase**
v2.6

**Scenario**
Industry

**To Reproduce**
1.Boot  ServiceVM  on ubuntu rootfs

2.Input "systemctl start docker" on ServiceVM

**Expected behavior**
docker starts successfully

**Actual result**

failed to start docker

Debugging info

root@intel-WL10:~# systemctl start docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.

root@intel-WL10:~# systemctl status docker
● docker.service - Docker Application Container Engine
Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
Drop-In: /etc/systemd/system/docker.service.d
└─50-runtime.conf, kata-containers.conf, proxy.conf
Active: failed (Result: exit-code) since Thu 2021-07-22 03:00:35 CST; 1min 10s ago
TriggeredBy: ● docker.socket
Docs: https://docs.docker.com
Process: 1322 ExecStart=/usr/bin/dockerd -D --add-runtime kata-runtime=/usr/bin/kata-runtime --default-runtime=kata-runtime (code=exited, statu>
Main PID: 1322 (code=exited, status=1/FAILURE)

Jul 22 03:00:35 intel-WL10 systemd[1]: docker.service: Scheduled restart job, restart counter is at 3.
Jul 22 03:00:35 intel-WL10 systemd[1]: Stopped Docker Application Container Engine.
Jul 22 03:00:35 intel-WL10 systemd[1]: docker.service: Start request repeated too quickly.
Jul 22 03:00:35 intel-WL10 systemd[1]: docker.service: Failed with result 'exit-code'.
Jul 22 03:00:35 intel-WL10 systemd[1]: Failed to start Docker Application Container Engine.

gvancuts, 2021-09-28:
Can you also post the `journalctl -xe -u docker` output?

GGGZZZRRR, 2021-09-29:
root@10239146151sos-dom0:~# systemctl restart docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
root@10239146151sos-dom0:~# journalctl -xe
Sep 29 06:35:43 10239146151sos-dom0 systemd[1]: Reached target Final Step.
-- Subject: A start job for unit final.target has finished successfully
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- A start job for unit final.target has finished successfully.
--
-- The job identifier is 1594.
Sep 29 06:35:43 10239146151sos-dom0 systemd[1]: systemd-poweroff.service: Succeeded.
-- Subject: Unit succeeded
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- The unit systemd-poweroff.service has successfully entered the 'dead' state.
Sep 29 06:35:43 10239146151sos-dom0 systemd[1]: Finished Power-Off.
-- Subject: A start job for unit systemd-poweroff.service has finished successfully
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- A start job for unit systemd-poweroff.service has finished successfully.
--
-- The job identifier is 1593.
Sep 29 06:35:43 10239146151sos-dom0 systemd[1]: Reached target Power-Off.
-- Subject: A start job for unit poweroff.target has finished successfully
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- A start job for unit poweroff.target has finished successfully.
--
-- The job identifier is 1592.
Sep 29 06:35:43 10239146151sos-dom0 systemd[1]: Shutting down.
Sep 29 06:35:43 10239146151sos-dom0 systemd-shutdown[1]: Syncing filesystems and block devices.
Sep 29 06:35:43 10239146151sos-dom0 systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Sep 29 06:35:43 10239146151sos-dom0 systemd-journald[309]: Journal stopped
-- Subject: The journal has been stopped
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- The system journal process has shut down and closed all currently
-- active journal files.

gvancuts, 2021-09-29:
My first suspect is some kernel options missing. I just checked the new-ish 5.10 that we use and ran the Moby check script on it. There are some options we'd need to turn on in it:
```
$ ./check-config.sh acrn-kernel/.config
info: reading kernel config from acrn-kernel/.config ...

Generally Necessary:
- cgroup hierarchy: properly mounted [/sys/fs/cgroup]
- apparmor: enabled and tools installed
- CONFIG_NAMESPACES: enabled
- CONFIG_NET_NS: enabled
- CONFIG_PID_NS: enabled
- CONFIG_IPC_NS: enabled
- CONFIG_UTS_NS: enabled
- CONFIG_CGROUPS: enabled
- CONFIG_CGROUP_CPUACCT: enabled
- CONFIG_CGROUP_DEVICE: enabled
- CONFIG_CGROUP_FREEZER: enabled
- CONFIG_CGROUP_SCHED: enabled
- CONFIG_CPUSETS: enabled
- CONFIG_MEMCG: enabled
- CONFIG_KEYS: enabled
- CONFIG_VETH: enabled
- CONFIG_BRIDGE: enabled
- CONFIG_BRIDGE_NETFILTER: missing
- CONFIG_IP_NF_FILTER: enabled
- CONFIG_IP_NF_TARGET_MASQUERADE: enabled
- CONFIG_NETFILTER_XT_MATCH_ADDRTYPE: enabled
- CONFIG_NETFILTER_XT_MATCH_CONNTRACK: enabled
- CONFIG_NETFILTER_XT_MATCH_IPVS: enabled
- CONFIG_NETFILTER_XT_MARK: enabled
- CONFIG_IP_NF_NAT: enabled
- CONFIG_NF_NAT: enabled
- CONFIG_POSIX_MQUEUE: enabled
- CONFIG_CGROUP_BPF: missing

Optional Features:
- CONFIG_USER_NS: enabled
- CONFIG_SECCOMP: enabled
- CONFIG_SECCOMP_FILTER: enabled
- CONFIG_CGROUP_PIDS: enabled
- CONFIG_MEMCG_SWAP: enabled
    (cgroup swap accounting is currently enabled)
- CONFIG_LEGACY_VSYSCALL_EMULATE: enabled
- CONFIG_BLK_CGROUP: enabled
- CONFIG_BLK_DEV_THROTTLING: enabled
- CONFIG_CGROUP_PERF: enabled
- CONFIG_CGROUP_HUGETLB: enabled
- CONFIG_NET_CLS_CGROUP: enabled
- CONFIG_CGROUP_NET_PRIO: enabled
- CONFIG_CFS_BANDWIDTH: enabled
- CONFIG_FAIR_GROUP_SCHED: enabled
- CONFIG_RT_GROUP_SCHED: enabled
- CONFIG_IP_NF_TARGET_REDIRECT: enabled
- CONFIG_IP_VS: enabled
- CONFIG_IP_VS_NFCT: enabled
- CONFIG_IP_VS_PROTO_TCP: enabled
- CONFIG_IP_VS_PROTO_UDP: enabled
- CONFIG_IP_VS_RR: enabled
- CONFIG_SECURITY_SELINUX: enabled
- CONFIG_SECURITY_APPARMOR: enabled
- CONFIG_EXT4_FS: enabled
- CONFIG_EXT4_FS_POSIX_ACL: enabled
- CONFIG_EXT4_FS_SECURITY: enabled
- Network Drivers:
  - "overlay":
    - CONFIG_VXLAN: missing
    - CONFIG_BRIDGE_VLAN_FILTERING: missing
      Optional (for encrypted networks):
      - CONFIG_CRYPTO: enabled
      - CONFIG_CRYPTO_AEAD: enabled
      - CONFIG_CRYPTO_GCM: enabled
      - CONFIG_CRYPTO_SEQIV: enabled
      - CONFIG_CRYPTO_GHASH: enabled
      - CONFIG_XFRM: enabled
      - CONFIG_XFRM_USER: enabled
      - CONFIG_XFRM_ALGO: enabled
      - CONFIG_INET_ESP: enabled
  - "ipvlan":
    - CONFIG_IPVLAN: missing
  - "macvlan":
    - CONFIG_MACVLAN: enabled
    - CONFIG_DUMMY: missing
  - "ftp,tftp client in container":
    - CONFIG_NF_NAT_FTP: enabled
    - CONFIG_NF_CONNTRACK_FTP: enabled
    - CONFIG_NF_NAT_TFTP: enabled
    - CONFIG_NF_CONNTRACK_TFTP: enabled
- Storage Drivers:
  - "aufs":
    - CONFIG_AUFS_FS: missing
      (note that some kernels include AUFS patches but not the AUFS_FS flag)
  - "btrfs":
    - CONFIG_BTRFS_FS: enabled
    - CONFIG_BTRFS_FS_POSIX_ACL: missing
  - "devicemapper":
    - CONFIG_BLK_DEV_DM: enabled
    - CONFIG_DM_THIN_PROVISIONING: enabled
  - "overlay":
    - CONFIG_OVERLAY_FS: missing
  - "zfs":
    - /dev/zfs: present
    - zfs command: missing
    - zpool command: missing

Limits:
- /proc/sys/kernel/keys/root_maxkeys: 1000000

```

fuzhongl, 2021-12-22:
[External_System_ID] ACRN-7549