projectacrn/acrn-hypervisor #6652: Refine system shutdown

wuxyintel, 2021-10-05:
Use socket(virtio-net) or vUART for communication between the host and guest side lifecycle managers. Need to update lifecycle manager. Lifecycle manager is a reference implementation, customer can implement more functions (such as shutdown request authority) according to their requirements.
Need to update hypervisor and config tool to support multi legacy vUART  and multi PCI vUART at the same time, the total number of vUART supported by hypervisor is  8.

yuchuyang, 2021-10-08:
[External_System_ID] ACRN-7407

NanlinXie, 2021-11-24:
Updated document:
https://projectacrn.github.io/latest/tutorials/enable_s5.html
https://projectacrn.github.io/latest/getting-started/getting-started.html

amyreye, 2022-02-17:
@NanlinXie @dbkinder Can we close this issue?

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie