projectacrn/acrn-hypervisor #6655: Launch Windows as the Guest VM, but there was nothing on the screen

LQ-lq, 2021-10-08:
I follow the tutorial https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html
when I run sudo ./install_win.sh, there was nothing on the screen.

~$ iasl -v
Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20210105
Copyright (c) 2000 - 2021 Intel Corporation

~$ acrn-dm -v
DM version is: 2.6-f5935afc4 (daily tag:acrn-2021w38.1-180000p), build by my@2021-09-30 03:15:45

~/work$ sudo ./install_win.sh
[sudo] password for tnhv5:
cpu1 online=1
cpu2 online=1
cpu3 online=1
cpu4 online=1
cpu5 online=1
cpu6 online=1
cpu7 online=1
./install_win.sh: line 14: echo: write error: No such device
SW_LOAD: get ovmf path /home/tnhv5/work/OVMF.fd, size 0x200000
vm_create: win_vm1
vm_setup_memory: size=0x100000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A                                                                                                                                                             18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A                                                                                                                                                             18B34643
level 0 free/need pages:0/1 page size:0x200000
level 1 free/need pages:12/4 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 1 > /sys/kernel/mm/hugepages/hugepages-2048kB/n                                                                                                                                                             r_hugepages
now enough free pages are reserved!

try to setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x80000000
total_size 0x1c0000000

mmap ptr 0x0x7f67e7fd5000 -> baseaddr 0x0x7f6800000000
mmap 0x80000000@0x7f6800000000
touch 2 pages with pagesz 0x40000000
mmap 0x80000000@0x7f6900000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7f68ffe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x80000000
vm_init_vdevs
No pm notify channel given
polling 34...
Listening 34...
init passthru first to reserve PIO BAR
init passthru first to reserve PIO BAR
pci init passthru
init passthru first to reserve PIO BAR
init passthru first to reserve PIO BAR
init passthru first to reserve PIO BAR
pci init passthru
init passthru first to reserve PIO BAR
pci init hostbridge
pci init lpc
pci init virtio-blk
pci init ahci
pci init ahci
pci init virtio-net
virtio_net_get_ifindex: Unable to get interface index on this platform
                                                                      tpm: init_                                                                                                                                                             vtpm2:Invalid socket path!
                          vm_get_config, l2_cat_shift=1, l3_cat_shift=4
                                                                       basl_fwri                                                                                                                                                             te_madt, dm_cpu_bitmask:0x0, hv_cpu_bitmask:0x3, guest_cpu_bitmask: 0x3
                                                                       basl_fwri                                                                                                                                                             te_madt, vcpu_id=0x0, pcpu_id=0x0, lapic_id=0x0
                                               basl_fwrite_madt, vcpu_id=0x1, pc                                                                                                                                                             pu_id=0x1, lapic_id=0x2
                       acrn_sw_load
                                   SW_LOAD: partition blob /home/tnhv5/work/OVMF                                                                                                                                                             .fd size 0x200000 copied to addr 0x7f68ffe00000
                                               SW_LOAD: build e820 6 entries to                                                                                                                                                              addr: 0x7f68000ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x0000000000100000, size 0x000000004b6fc000,  type 0x1
SW_LOAD: entry[2]: addr 0x000000004b7fc000, size 0x0000000004004000,  type 0x2
SW_LOAD: entry[3]: addr 0x000000004f800000, size 0x0000000030800000,  type 0x1
SW_LOAD: entry[4]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[5]: addr 0x0000000100000000, size 0x0000000080000000,  type 0x1
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu

LQ-lq, 2021-10-21:
> Do you refer following guide to build OVMF.fd with GVT-d GOP feature? https://projectacrn.github.io/latest/tutorials/gpu-passthru.html

Yes, I followed this guide and generated OVMF.fd

LQ-lq, 2021-10-21:
> It looks like there is something wrong with the bind/unbind operations. Can you provide the output of `lspci` on your platform?

:~$ lspci
00:00.0 Host bridge: Intel Corporation Device 9a14 (rev 01)
00:02.0 VGA compatible controller: Intel Corporation Device 9a49 (rev 01)
00:06.0 PCI bridge: Intel Corporation Device 9a09 (rev 01)
00:07.0 PCI bridge: Intel Corporation Device 9a25 (rev 01)
00:07.2 PCI bridge: Intel Corporation Device 9a27 (rev 01)
00:08.0 System peripheral: Intel Corporation Device 9a11 (rev 01)
00:0d.0 USB controller: Intel Corporation Device 9a13 (rev 01)
00:0d.2 USB controller: Intel Corporation Device 9a1b (rev 01)
00:0d.3 USB controller: Intel Corporation Device 9a1d (rev 01)
00:14.0 USB controller: Intel Corporation Device a0ed (rev 20)
00:14.2 RAM memory: Intel Corporation Device a0ef (rev 20)
00:14.3 Network controller: Intel Corporation Device a0f0 (rev 20)
00:15.0 Serial bus controller [0c80]: Intel Corporation Device a0e8 (rev 20)
00:15.1 Serial bus controller [0c80]: Intel Corporation Device a0e9 (rev 20)
00:16.0 Communication controller: Intel Corporation Device a0e0 (rev 20)
00:16.3 Serial controller: Intel Corporation Device a0e3 (rev 20)
00:17.0 SATA controller: Intel Corporation Device a0d3 (rev 20)
00:1d.0 PCI bridge: Intel Corporation Device a0b1 (rev 20)
00:1f.0 ISA bridge: Intel Corporation Device a082 (rev 20)
00:1f.3 Audio device: Intel Corporation Device a0c8 (rev 20)
00:1f.4 SMBus: Intel Corporation Device a0a3 (rev 20)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device a0a4 (rev 20)
01:00.0 Non-Volatile memory controller: Device 1d97:2263 (rev 03)
58:00.0 Ethernet controller: Intel Corporation Device 15f2 (rev 03)

fuzhongl, 2022-07-05:
> > > Can you try this `install_win.sh` script to see if it helps?
> > 
> > 
> > The screen is still black without any display.
> 
> @fuzhongl , this is a Tiger Lake (TGL) platform, do you know if we have ever tested this on it? The GVT-d (GOP) tutorial does not mention it.

@gvancuts Sorry for not aware this message.
Yes, GVT-d is verified on TGLNUC and TGL vecow with v3.0

closed 2022-07-05 by fuzhongl