projectacrn/acrn-hypervisor #6689: [DX]Board Inspector tool error message refinement

NanlinXie, 2021-10-19:
From DX survey, user was confused for the output message from board inspector, we need refine

Change Board Inspector tool success message to be clear that users were successful creating the board configuration file. Success message: “my_board.xml saved successfully!”
Board Inspector: Suppress warnings if they do not require user action.
Board Inspector: Ensure error messages are clear about the exact issue and provide a resolution to users.

NanlinXie, 2021-10-19:
[External_System_ID] ACRN-7430

amyreye, 2022-03-21:
@NanlinXie Is the Engineering team ready to update docs for this issue? For example, in ``board_inspector.py``, it looks like the log levels were updated in #7116; however, the command-line help wasn't updated and docs were not updated. For example, ``board_inspector.py`` command-line help only mentions 3 log levels, but code has 5 levels:

``
    parser.add_argument("--loglevel", default="warning", help="choose log level, e.g. info, warning or error")

NanlinXie, 2022-03-23:
@amyreye , yes the doc need to be updated, but for now, we are still ongoing coding work.