projectacrn/acrn-hypervisor #6826: doc: Review Supported Hardware

amyreye, 2021-11-12:
**Describe the bug**
Review the Supported Hardware document and update as needed to reflect current support.
https://projectacrn.github.io/latest/reference/hardware.html

dbkinder, 2021-11-12:
The supported hardware document needs to inform developers what hardware platform they can use to work with ACRN, and what ACRN capabilities are (and aren't) available on specific platforms.  While we should recommend and verify a specific platform for open source developers with an ACRN release, we might need to rely on the board inspector and configurator to be the authority about other platforms.  Specifically, these tools should clearly report any issues about the presented platform hardware when probing for board information (with the board inspector) or when trying to configure a feature (in the configurator) that the board doesn't support.

amyreye, 2021-12-16:
Addressed in PR #6971

closed 2021-12-16 by amyreye