projectacrn/acrn-hypervisor #6840: I'm fallowing

Waiting-For-Wind, 2021-11-15:
**Describe the bug**
A clear and concise description of what the bug is.

**Platform**
Describe the platform information, 

**Codebase**
Describe which codebase is used.

**Scenario**
Describe the scenario: Hybrid, Industry or logical_partition

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Attach the following logs if available:

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

gvancuts, 2021-11-15:
hi @Waiting-For-Wind , we need more details on your set-up. Can you start by describing it?
- HW environment
- ACRN version

Thanks!

gvancuts, 2021-11-15:
> Your response speed is amazing，and I'm sorry , when I first Edit this my browser crashed. I'm so sorry.

No worries at all! ;-)

gvancuts, 2022-03-02:
Hi @Waiting-For-Wind , is this still an issue?

closed 2022-04-07 by Waiting-For-Wind