projectacrn/acrn-hypervisor #6858: Resource Allocation is not supported!

Waiting-For-Wind, 2021-11-18:
When I run `board_inspector.py` The terminal print :
```
Resource Allocation is not supported!
WARNING:root:Cannot read the configuration space of 00:14.1 from /sys/devices/pci0000:00/0000:00:14.1/config. Assume the PCI device does not exist.
```

I want  know ,  is this WARRNING or `Resource Allocation is not supported!` fatal for installing ACRN?

gvancuts, 2021-11-18:
> Resource Allocation is not supported!

These are two separate messages. The first one indicates your HW (processor) does not support Resoure Allocation. This is not a problem to run ACRN, these are HW features that help tweak and improve the real-time behaviour of certain VMs. This is an umbrella term. You can read more about it here: https://www.intel.com/content/www/us/en/architecture-and-technology/resource-director-technology.html

> WARNING:root:Cannot read the configuration space of 00:14.1 from /sys/devices/pci0000:00/0000:00:14.1/config. Assume the PCI device does not exist.

What device is this? Can you paste the output of `sudo lspci -vvv -s 00:14.1`?

fuzhongl, 2022-07-05:
Close it since no feedback long time.

closed 2022-07-05 by fuzhongl