projectacrn/acrn-hypervisor #6902: Board inspector fails after enabling NTB

villanyibalint, 2021-11-25:
**Describe the bug**
Board inspector crashes with sysfs hierarchies where a leaf matched by the BDF regex lacks a config file.

I encountered this situation after enabling the Non-Transparent Bridge capability in my Skylake Scalable based platform. Linux generates a sysfs hierarchy that resembles the following:

- sys
  - devices
    - pci0000:85
      - 0000:85:00.0
        - 0000:85:00.0
          - _no config_
        - config
        - other files ...
      - other devices on bus 85 ...

Since the device [parser recursively checks](https://github.com/projectacrn/acrn-hypervisor/blob/v2.6/misc/config_tools/board_inspector/extractors/60-pci.py#L179) each leaf for new devices, the second _0000:85:00.0_ folder is also regarded as a PCI device, but the script crashes while trying to [access the PCI config space](https://github.com/projectacrn/acrn-hypervisor/blob/v2.6/misc/config_tools/board_inspector/extractors/60-pci.py#L69), since this second folder does not have such a file.

I have no idea why Linux generates such a hierarchy and whether this happens only with an NTB PCI device.

**Platform**
Intel Skylake Xeon Scalable based x86 platform
Ubuntu 5.4.0-90-generic

**Codebase**
v2.6 tag

**Scenario**
Not specific to scenario

**To Reproduce**
1. Configure a Skylake core NTB in back-to-back mode (possibly works with non-back-to-back NTB mode as well)
2. Run the board inspector script
3. Script fails without producing an output

**Expected behavior**
The script does not fail and produces a board config. Possibly ignores the leaf without a config file?

gvancuts, 2021-11-29:
@NanlinXie , I don't have the ability to test this. Can you look for someone to take a loot?

junjiemao1, 2021-11-30:
Thanks for reporting this issue. We have not seen such hierarchy before and still need some time to investigate what's really happening inside the kernel. Meanwhile, I have drafted a work-around [patch](https://github.com/junjiemao1/acrn-hypervisor/commit/d175a8453822a4d9891c39c28bac9ccebfc0fefe) to at least skip these ill-formed directories.

Unfortunately we do not have a platform with NTB yet. Is it possible for you to try applying this patch to your local repo and verify if it works as expected? Thanks!

villanyibalint, 2021-12-02:
Hi, yes I'll try the patch locally, as soon as I can

NanlinXie, 2022-01-16:
@junjiemao1 , do we need upstream this WR patch to avoid board inspector crash?

junjiemao1, 2022-01-17:
We can do that. I'll submit the patches for review in the mailing list then.

closed 2022-07-05 by fuzhongl