projectacrn/acrn-hypervisor #6931: RTVM CPUID 0x2 EBX value is not equal to HV cpuid 0x2 EBX

lirui34, 2021-12-01:
**Describe the bug**
RTVM CPUID 0x2 EBX value is not equal to HV cpuid 0x2 EBX

**Codebase**
release_2.7 branch

**Scenario**
shared

**To Reproduce**
Steps to reproduce the behavior:
1. boot shared scenario and then launch post rtvm
2. run cpuid -l 0x2 -r  cmd in rtvm
3. run cpuid 0x2 in HV console

**Expected behavior**
result should be the same value.

**Actual result**
02H.EBX value is not the same:

```
CPU 0:
   0x00000002 0x00: eax=0x00feff01 ebx=0x00000000 ecx=0x00000000 edx=0x00000000
CPU 1:
   0x00000002 0x00: eax=0x00feff01 ebx=0x00000000 ecx=0x00000000 edx=0x00000000
root@intel-corei7-64:~#

 ---Entering ACRN SHELL---
ACRN:\>cpuid 0x2
cpuid leaf: 0x2, subleaf: 0x0, 0xfeff01:0xf0:0x0:0x0
```

***Reproduce rate***:
_100%_

lirui34, 2021-12-01:
[External_System_ID] ACRN-7517

NanlinXie, 2022-01-15:
Patch merged and close with no regression.

closed 2022-01-15 by NanlinXie