projectacrn/acrn-hypervisor #6950: Cannot launch hard RT user VM with Ubuntu 20.04.3 and preempt-RT kernel 5.10.52-rt47

christophsusen, 2021-12-06:
**Describe the bug**
Cannot launch hard RT user VM with Ubuntu 20.04.3 and preempt-RT kernel 5.10.52-rt47. The user VM gets stuck in the starting process when using the launch script for an RT-VM. When a launch script for a non-RT-VM is used with the same image, the VM starts successfully.

**Platform**
Dell OptiPlex 5070 with Intel Core i7-9700, 8 GB RAM and 256 GB NVME SSD. The board configuration file can be found [here](https://github.com/projectacrn/acrn-hypervisor/files/7659012/board_configuration.txt).

**Codebase**
ACRN kernel v2.6
ACRN hypervisor v2.6
SOS Ubuntu 20.04.3
UOS Ubuntu 20.04.3 with preempt-RT kernel 5.10.52-rt47

**Scenario**
Custom scenario derived from the industry scenario. Scenario contains one non-RT-VM (VM number one) and one RT-VM (VM number two). These VMs are basically identical to VMs three and two from the industry scenario.

**To Reproduce**
Steps to reproduce the behavior:
1. Start user VM with Ubuntu 20.04.3 and preempt-RT kernel 5.10.52-rt47 using the launch script for the RT-VM in the industry scenario.

**Expected Behavior**
From my point of understanding, my configuration should work and the RT user VM should start. If anything with my configuration is wrong, please let me know.

**Additional Context**
- [Board configuration file](https://github.com/projectacrn/acrn-hypervisor/files/7659012/board_configuration.txt)
- [Scenario configuration file](https://github.com/projectacrn/acrn-hypervisor/files/7659011/scenario_configuration.txt)
- [Launch script non-RT-VM](https://github.com/projectacrn/acrn-hypervisor/files/7658925/launch_uos_id1.txt)
- [Launch script RT-VM](https://github.com/projectacrn/acrn-hypervisor/files/7658926/launch_uos_id2.txt)
- [Grub menu entry for user VM](https://github.com/projectacrn/acrn-hypervisor/files/7658962/grub_menu.txt)
- [Log from /var/log/acrn-dm](https://github.com/projectacrn/acrn-hypervisor/files/7659054/acrn_dm_log.txt)
- [Serial port log (ACRN shell and vm_console)](https://github.com/projectacrn/acrn-hypervisor/files/7659027/serial_port_log.txt)
- [Console log when running launch script for RT-VM](https://github.com/projectacrn/acrn-hypervisor/files/7659020/console_log.txt)

Thanks in advance for your help!

_I changed the file ending to .txt for all uploaded files since GitHub does not accept .xml and .sh files._

fuzhongl, 2021-12-07:
@christophsusen 
Is CONFIG_VIRTIO_PMD enabled in your RT-kernel config?
If not, Please refer following patch: 
https://github.com/projectacrn/acrn-kernel/commit/0e8ea675e40db608646417fe99c0db32c5f6dd28

thanks!

christophsusen, 2021-12-07:
Hi @fuzhongl, thanks for your reply!

Are you referring to the kernel config for the ACRN service VM or to the kernel config for the Ubuntu user VM?
In none of these, I can find a parameter named CONFIG_VIRTIO_PMD.

Best regards,
Christoph

fuzhongl, 2021-12-08:
Usually, guest_flag isn't needed for post VM:
Could you help to rebuild acrn with  following change in scenario.xml? 
```
  <vm id="2">
    <vm_type>POST_RT_VM</vm_type>
    <guest_flags>
      <guest_flag>GUEST_FLAG_LAPIC_PASSTHROUGH</guest_flag>
      <guest_flag>GUEST_FLAG_RT</guest_flag>
    </guest_flags> 
``` 

To

```
  <vm id="2">
    <vm_type>POST_RT_VM</vm_type>
    <guest_flags>
      <guest_flag>0</guest_flag>
    </guest_flags> 
```

BR.
Fuzhong

christophsusen, 2021-12-08:
I made the changes to the scenario configuration and rebuild the hypervisor. Unfortunately, the start-up process of the VM is still awfully slow.

Thanks for your help!

Best regards,
Christoph

christophsusen, 2021-12-10:
Since I read at https://projectacrn.github.io/latest/tutorials/rtvm_performance_tips.html?highlight=lapic_pt#mandatory-options-for-an-rtvm that `--lapic_pt` is mandatory for RTVMs, I would still say that my issue is not resolved. Do you have any further ideas on what the problem could be?

fuzhongl, 2021-12-13:
Could you please disable Hyper-thread in BIOS and re-generate the board.xml? 
Then rebuild acrn.bin with the updated board.xml.

Thanks!

BR.
Fuzhong

christophsusen, 2021-12-13:
Hi @fuzhongl, thanks for your answer!

I don't think that my CPU even supports hyper-threading. Here is the output that I get from `lscpu`:
```
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   39 bits physical, 48 bits virtual
CPU(s):                          8
On-line CPU(s) list:             0-7
Thread(s) per core:              1
Core(s) per socket:              8
Socket(s):                       1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           158
Model name:                      Intel(R) Core(TM) i7-9700 CPU @ 3.00GHz
Stepping:                        13
CPU MHz:                         3000.000
CPU max MHz:                     3000.0000
CPU min MHz:                     800.0000
BogoMIPS:                        6000.00
L1d cache:                       256 KiB
L1i cache:                       256 KiB
L2 cache:                        2 MiB
L3 cache:                        12 MiB
Vulnerability Itlb multihit:     Processor vulnerable
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Enhanced IBRS, IBPB conditional, RSB filling
Vulnerability Srbds:             Mitigation; TSX disabled
Vulnerability Tsx async abort:   Mitigation; TSX disabled
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc rep_good nopl xtopolo
                                 gy nonstop_tsc cpuid aperfmperf pni pclmulqdq monitor est tm2 ssse3 fma cx16 xtpr pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor la
                                 hf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt xsaveopt
                                 xsavec xgetbv1 xsaves dtherm arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities
```

Best regards,
Christoph

fuzhongl, 2021-12-13:
Could  you please input `vcpu_list` and `vm_console 2` in HV console with `--lapic_pt` in launch script?

Thanks!

BR.
Fuzhong

christophsusen, 2021-12-13:
This is the output of `vcpu_list`:
```
VM ID    PCPU ID    VCPU ID    VCPU ROLE    VCPU STATE    THREAD STATE
=====    =======    =======    =========    ==========    ==========
  0         0          0       PRIMARY      Running          BLOCKED
  0         1          1       SECONDARY    Running          BLOCKED
  0         2          2       SECONDARY    Running          BLOCKED
  0         3          3       SECONDARY    Running          BLOCKED
  2         4          0       PRIMARY      Running          RUNNING
  2         5          1       SECONDARY    Running          RUNNING
  2         6          2       SECONDARY    Running          RUNNING
  2         7          3       SECONDARY    Running          RUNNING
```

Please note that I changed the number of CPUs for the RTVM to 4 while experimenting with the problem.

This is the output I get when running `vm_console 2`:
```
[    0.000000] Linux version 5.10.52-rt47 (ssn@ssn-optiplex-5070-uos-3) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1 SMP PREEMPT_RT Tue Dec 7 16:44:17 CET 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.10.52-rt47 root=UUID=74f2a1a0-ad7e-4bda-a851-f3efe55cb507 ro rootwait nohpet console=hvc0 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll irqaffinity=0
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007e7d1fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007e7d2000-0x000000007e7d2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007e7d3000-0x000000007e7f4fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007e7f5000-0x000000007e91afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007e91b000-0x000000007f99afff] usable
[    0.000000] BIOS-e820: [mem 0x000000007f99b000-0x000000007f9eefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007f9ef000-0x000000007f9f3fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007f9f4000-0x000000007f9fafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007f9fb000-0x000000007f9fefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007f9ff000-0x000000007ffdffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffe0000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffe00000-0x00000000ffe1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] printk: debug: ignoring loglevel setting.
[    0.000000] process: using polling idle threads
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by ARCN
[    0.000000] efi: ACPI=0x7f9fa000 ACPI 2.0=0x7f9fa014 MEMATTR=0x7f559118 MOKvar=0x7f45f000
[    0.000000] DMI not present or invalid.
[    0.000000] Hypervisor detected: ACRN
[    0.000000] tsc: Detected 3000.000 MHz processor
[    0.000010] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000012] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000017] last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.000037] MTRR default type: uncachable
[    0.000038] MTRR fixed ranges enabled:
[    0.000039]   00000-FFFFF write-back
[    0.000040] MTRR variable ranges enabled:
[    0.000169] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.000669] CPU MTRRs all blank - virtualized system.
[    0.000674] last_pfn = 0x7ffe0 max_arch_pfn = 0x400000000
[    0.000742] found SMP MP-table at [mem 0x000f0000-0x000f000f]
[    0.000759] check: Scanning 1 areas for low memory corruption
[    0.000819] Using GB pages for direct mapping
[    0.008658] printk: log_buf_len: 16777216 bytes
[    0.008660] printk: early log buf free: 258736(98%)
[... 372 lines cut ...]
[    2.603835] virtio_blk virtio0: [vda] 104857600 512-byte logical blocks (53.7 GB/50.0 GiB)
[    2.603839] vda: detected capacity change from 0 to 53687091200
[    2.603907] virtio_blk virtio0: start polling timer: 10000000
[    3.292834] virtio_net virtio2: start polling timer: 10000000
[    3.293146] libphy: r8169: probed
[    3.305549] r8169 0000:00:04.0 eth1: RTL8168e/8111e, 3c:49:37:07:07:11, XID 2c2, IRQ 84
[    3.305553] r8169 0000:00:04.0 eth1: jumbo features [frames: 9194 bytes, tx checksumming: ko]
[    3.436116] virtio_net virtio2 enp0s9: renamed from eth0
[    3.454996] r8169 0000:00:04.0 enp0s4: renamed from eth1
[    4.284802]  vda: vda1 vda2
[    5.565435] EXT4-fs (vda2): mounted filesystem with ordered data mode. Opts: (null)
[   38.293042] systemd[1]: Inserted module 'autofs4'
[... cut ...]
[   38.582082] systemd[1]: Detected virtualization acrn.
[   38.582085] systemd[1]: Detected architecture x86-64.
[   68.627883] systemd[1]: Set hostname to <ssn-optiplex-5070-uos-3>.
[  101.756394] systemd[1]: Created slice system-modprobe.slice.
[  132.111189] systemd[1]: Created slice system-serial\x2dgetty.slice.
```

The VM is not completely stuck at that point. There are new messages being printed really slowly. However, I never had the VM start completely even when I waited very long.

Best regards,
Christoph

fuzhongl, 2021-12-13:
@mcao6 Could you please help to check the boot issue for RTVM?
Thanks!

BR.
Fuzhong

mgcao, 2021-12-13:
@christophsusen 
just checked the history of this issue. it is a strange issue, please help to check following:

1. add no_ipi_broadcast=1 to your RTVM bootargs, and boot to check.
2. if still slow, just as your test, remove --lapic_pt, and boot to put the "dmesg" of VM2 info here, let me compare.

from the log:
[    5.565435] EXT4-fs (vda2): mounted filesystem with ordered data mode. Opts: (null)
[   38.293042] systemd[1]: Inserted module 'autofs4'

the time seems spending on FS module. but not sure why.

christophsusen, 2021-12-13:
Hi @mgcao, thanks for your quick answer!

I added `no_ipi_broadcast=1` to the kernel command line of the RTVM. Unfortunately, that did not change the situation.

Afterwards, I removed `--lapic_pt` from the launch script and added `--rtvm` to obtain the following `dmesg` output:
```
[    0.000000] Linux version 5.10.52-rt47 (ssn@ssn-optiplex-5070-uos-3) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1 SMP PREEMPT_RT Tue Dec 7 16:44:17 CET 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.10.52-rt47 root=UUID=74f2a1a0-ad7e-4bda-a851-f3efe55cb507 ro rootwait nohpet console=hvc0 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll irqaffinity=0 no_ipi_broadcast=1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007e7d1fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007e7d2000-0x000000007e7d2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007e7d3000-0x000000007e7f4fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007e7f5000-0x000000007e91afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007e91b000-0x000000007f99afff] usable
[    0.000000] BIOS-e820: [mem 0x000000007f99b000-0x000000007f9eefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007f9ef000-0x000000007f9f3fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007f9f4000-0x000000007f9fafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007f9fb000-0x000000007f9fefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007f9ff000-0x000000007ffdffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffe0000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffe00000-0x00000000ffe1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] printk: debug: ignoring loglevel setting.
[    0.000000] process: using polling idle threads
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by ARCN
[    0.000000] efi: ACPI=0x7f9fa000 ACPI 2.0=0x7f9fa014 MEMATTR=0x7f559118 MOKvar=0x7f45f000 
[    0.000000] DMI not present or invalid.
[    0.000000] Hypervisor detected: ACRN
[    0.000000] tsc: Detected 3000.000 MHz processor
[    0.000010] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000012] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000017] last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.000037] MTRR default type: uncachable
[    0.000038] MTRR fixed ranges enabled:
[    0.000039]   00000-FFFFF write-back
[    0.000040] MTRR variable ranges enabled:
[    0.000168] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000665] CPU MTRRs all blank - virtualized system.
[    0.000670] last_pfn = 0x7ffe0 max_arch_pfn = 0x400000000
[    0.000738] found SMP MP-table at [mem 0x000f0000-0x000f000f]
[    0.000755] check: Scanning 1 areas for low memory corruption
[    0.000812] Using GB pages for direct mapping
[    0.008893] printk: log_buf_len: 16777216 bytes
[    0.008895] printk: early log buf free: 258720(98%)
[    0.008896] Secure boot disabled
[    0.008897] RAMDISK: [mem 0x3a30d000-0x3d060fff]
[    0.008901] ACPI: Early table checksum verification disabled
[    0.008906] ACPI: RSDP 0x000000007F9FA014 000024 (v02 DM    )
[    0.008910] ACPI: XSDT 0x000000007F9F90E8 000044 (v01 DM     DMFACP   00000001      01000013)
[    0.008916] ACPI: FACP 0x000000007F9F7000 00010C (v05 DM     DMFACP   00000001 INTL 20210105)
[    0.008923] ACPI: DSDT 0x000000007F9F4000 0008A2 (v02 DM     DMDSDT   00000001 INTL 20210105)
[    0.008929] ACPI: FACS 0x000000007F9FE000 000040
[    0.008932] ACPI: APIC 0x000000007F9F8000 00005E (v01 DM     DMMADT   00000001 INTL 20210105)
[    0.008935] ACPI: HPET 0x000000007F9F6000 000038 (v01 DM     DMHPET   00000001 INTL 20210105)
[    0.008939] ACPI: MCFG 0x000000007F9F5000 00003C (v01 DM     DMMCFG   00000001 INTL 20210105)
[    0.008941] ACPI: Reserving FACP table memory at [mem 0x7f9f7000-0x7f9f710b]
[... 453 lines cut ...]
[   17.868605] systemd[1]: Condition check resulted in Rebuild Hardware Database being skipped.
[   17.868651] systemd[1]: Condition check resulted in Platform Persistent Storage Archival being skipped.
[   17.869535] systemd[1]: Starting Load/Save Random Seed...
[   17.873738] ppdev: user-space parallel port driver
[   17.877883] systemd[1]: Starting Create System Users...
[   17.920297] systemd[1]: Started Journal Service.
[... cut ...]
[   21.226294] intel_rapl_common: Found RAPL domain core
[   21.226295] intel_rapl_common: Found RAPL domain uncore
[   21.226296] intel_rapl_common: Found RAPL domain dram
[   27.594006] RTL8211DN Gigabit Ethernet r8169-20:00: attached PHY driver [RTL8211DN Gigabit Ethernet] (mii_bus:phy_addr=r8169-20:00, irq=IGNORE)
[   27.894186] r8169 0000:00:04.0 enp0s4: Link is Down
[   30.154205] r8169 0000:00:04.0 enp0s4: Link is Up - 1Gbps/Full - flow control rx/tx
[   30.154219] IPv6: ADDRCONF(NETDEV_CHANGE): enp0s4: link becomes ready
```

mgcao, 2021-12-14:
from the comparison, I doubt two points:
1. filesystem, it seems it is very slow for root fs mount, not sure why.
2. network, you pass through a network? r8169 0000:00:04.0 

please attach your launch RTVM script here; 
and lspci in SOS

mgcao, 2021-12-14:
also you can： ps -aux | grep dm
in SOS, send the info

christophsusen, 2021-12-14:
Hi @mgcao, I'm sorry, yes, I added a passthrough NIC while experimenting and also tried using a passthrough SATA disk for the RTVM instead of virtio-blk. However, that did not change anything. The non-RTVM boots fine with passthrough NIC and SATA but the RTVM still has the same problems.

Here is the launch script used for the RTVM to generate the `dmesg` output in my last comment (currently without `--lapic_pt`):
```
#!/bin/bash
# board: DELL_OPTIPLEX_5070, scenario: INDUSTRY, uos: PREEMPT-RT LINUX
# pci devices for passthru
declare -A passthru_vpid
declare -A passthru_bdf

passthru_vpid=(
["sata"]="8086 a352"
["ethernet"]="10ec 8168"
)
passthru_bdf=(
["sata"]="0000:00:17.0"
["ethernet"]="0000:01:00.0"
)

function tap_net() {
# create a unique tap device for each VM
tap=$1
tap_exist=$(ip a | grep "$tap" | awk '{print $1}')
if [ "$tap_exist"x != "x" ]; then
  echo "tap device existed, reuse $tap"
else
  ip tuntap add dev $tap mode tap
fi

# if acrn-br0 exists, add VM's unique tap device under it
br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
  echo "acrn-br0 bridge aleady exists, adding new tap device to it..."
  ip link set "$tap" master acrn-br0
  ip link set dev "$tap" down
  ip link set dev "$tap" up
fi
}

function launch_preempt_rt_linux()
{
#vm-name used to generate uos-mac address
mac=$(cat /sys/class/net/e*/address)
vm_name=post_vm_id$1
mac_seed=${mac:0:17}-${vm_name}

tap_net tap_tap_uos2
#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep -w "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

modprobe pci_stub
# Passthrough SATA
#echo ${passthru_vpid["sata"]} > /sys/bus/pci/drivers/pci-stub/new_id
#echo ${passthru_bdf["sata"]} > /sys/bus/pci/devices/${passthru_bdf["sata"]}/driver/unbind
#echo ${passthru_bdf["sata"]} > /sys/bus/pci/drivers/pci-stub/bind

# Passthrough ETHERNET
echo ${passthru_vpid["ethernet"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["ethernet"]} > /sys/bus/pci/devices/${passthru_bdf["ethernet"]}/driver/unbind
echo ${passthru_bdf["ethernet"]} > /sys/bus/pci/drivers/pci-stub/bind

mem_size=4096M
#interrupt storm monitor for pass-through devices, params order:
#threshold/s,probe-period(s),intr-inject-delay-time(ms),delay-duration(ms)
intr_storm_monitor="--intr_monitor 10000,10,1,100"
[... 55 lines cut ...]
00:14.2 RAM memory: Intel Corporation Cannon Lake PCH Shared SRAM (rev 10)
00:15.0 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #0 (rev 10)
00:16.0 Communication controller: Intel Corporation Cannon Lake PCH HECI Controller (rev 10)
00:16.3 Serial controller: Intel Corporation Cannon Lake PCH Active Management Technology - SOL (rev 10)
00:17.0 SATA controller: Intel Corporation Cannon Lake PCH SATA AHCI Controller (rev 10)
00:1b.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #17 (rev f0)
00:1c.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #4 (rev f0)
00:1c.6 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #7 (rev f0)
00:1d.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #9 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Q370 Chipset LPC/eSPI Controller (rev 10)
00:1f.3 Audio device: Intel Corporation Cannon Lake PCH cAVS (rev 10)
00:1f.4 SMBus: Intel Corporation Cannon Lake PCH SMBus Controller (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH SPI Controller (rev 10)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (7) I219-V (rev 10)
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 06)
02:00.0 Non-Volatile memory controller: SK hynix Device 1627
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 06)
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 06)
05:00.0 Serial controller: MosChip Semiconductor Technology Ltd. PCIe 9912 Multi-I/O Controller
```

And, finally, the output of `ps -aux | grep acrn-dm` from within the service VM after starting the RTVM (please let me know if that is not what you wanted):
`root        4072  3.5  0.0 7586664 6268 pts/0    Sl+  08:08   0:01 acrn-dm -A -m 4096M -s 0:0,hostbridge -U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 --mac_seed e4:54:e8:a3:33:30-post_vm_id2 --lapic_pt --virtio_poll 1000000 --logger_setting console,level=4;kmsg,level=3;disk,level=5 -s 6,virtio-blk,/home/acrn/acrn-target/acrn-work/images/ubuntu/uos-3-50g.img -s 9,virtio-net,tap_tap_uos2 -s 8,virtio-console,@stdio:stdio_port --ovmf /usr/share/acrn/bios/OVMF.fd --intr_monitor 10000,10,1,100 -s 4,passthru,01/00/0 post_vm_id2`

Thanks a lot for helping me!

mgcao, 2021-12-14:
to be honest, this is very wired issue, we need more try and log info:
1. just remove :      $intr_storm_monitor
2.  remove:   -s 4,passthru,01/00/0 \  

More log info:
3. offer the dm log in SOS: /var/log/acrn-dm/ 

4. is there some exception info in ACRN hypervisor console?  (the console you input vcpu_list)
5. maybe we need vmexit: https://github.com/projectacrn/acrn-hypervisor/commit/8ae4c90747edad7828e8b24328616b65b3265cba
 the patch is in release_2.5, you can port to your build, not too much work.

in ACRN console before you launch: vmexit clear 
after launch (or in the stuck): vmexit
offer the vmexit info.

christophsusen, 2021-12-14:
I removed the lines you mentioned from the launch script. The new launch script looks like this:
```
#!/bin/bash
# board: DELL_OPTIPLEX_5070, scenario: INDUSTRY, uos: PREEMPT-RT LINUX
# pci devices for passthru
declare -A passthru_vpid
declare -A passthru_bdf

passthru_vpid=(
["sata"]="8086 a352"
["ethernet"]="10ec 8168"
)
passthru_bdf=(
["sata"]="0000:00:17.0"
["ethernet"]="0000:01:00.0"
)

function tap_net() {
# create a unique tap device for each VM
tap=$1
tap_exist=$(ip a | grep "$tap" | awk '{print $1}')
if [ "$tap_exist"x != "x" ]; then
  echo "tap device existed, reuse $tap"
else
  ip tuntap add dev $tap mode tap
fi

# if acrn-br0 exists, add VM's unique tap device under it
br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
  echo "acrn-br0 bridge aleady exists, adding new tap device to it..."
  ip link set "$tap" master acrn-br0
  ip link set dev "$tap" down
  ip link set dev "$tap" up
fi
}

function launch_preempt_rt_linux()
{
#vm-name used to generate uos-mac address
mac=$(cat /sys/class/net/e*/address)
vm_name=post_vm_id$1
mac_seed=${mac:0:17}-${vm_name}

tap_net tap_tap_uos2
#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep -w "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

modprobe pci_stub
# Passthrough SATA
#echo ${passthru_vpid["sata"]} > /sys/bus/pci/drivers/pci-stub/new_id
#echo ${passthru_bdf["sata"]} > /sys/bus/pci/devices/${passthru_bdf["sata"]}/driver/unbind
#echo ${passthru_bdf["sata"]} > /sys/bus/pci/drivers/pci-stub/bind

# Passthrough ETHERNET
#echo ${passthru_vpid["ethernet"]} > /sys/bus/pci/drivers/pci-stub/new_id
#echo ${passthru_bdf["ethernet"]} > /sys/bus/pci/devices/${passthru_bdf["ethernet"]}/driver/unbind
#echo ${passthru_bdf["ethernet"]} > /sys/bus/pci/drivers/pci-stub/bind

mem_size=4096M
#interrupt storm monitor for pass-through devices, params order:
#threshold/s,probe-period(s),intr-inject-delay-time(ms),delay-duration(ms)
intr_storm_monitor="--intr_monitor 10000,10,1,100"

#logger_setting, format: logger_name,level; like following
[... 84 lines cut ...]
[2021-12-14 08:57:51][ 2704.913991] touch 1 pages with pagesz 0x200000
[2021-12-14 08:57:52][ 2705.336095] init passthru first to reserve PIO BAR
[2021-12-14 08:57:52][ 2705.336573] pci init virtio-net^M
[2021-12-14 08:57:52][ 2705.347460] basl_fwrite_madt, dm_cpu_bitmask:0x0, hv_cpu_bitmask:0xf0, guest_cpu_bitmask: 0xf0
[2021-12-14 08:57:52][ 2705.347473] basl_fwrite_madt, vcpu_id=0x0, pcpu_id=0x4, lapic_id=0x8
[2021-12-14 08:57:52][ 2705.347483] basl_fwrite_madt, vcpu_id=0x1, pcpu_id=0x5, lapic_id=0xa
[2021-12-14 08:57:52][ 2705.347493] basl_fwrite_madt, vcpu_id=0x2, pcpu_id=0x6, lapic_id=0xc
[2021-12-14 08:57:52][ 2705.347502] basl_fwrite_madt, vcpu_id=0x3, pcpu_id=0x7, lapic_id=0xe
[2021-12-14 08:57:52][ 2705.365683] acrn_sw_load
[2021-12-14 08:57:52][ 2705.368444] SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 0x200000 copied to addr 0x7fb63fe00000
[2021-12-14 08:57:52][ 2705.368461] SW_LOAD: build e820 4 entries to addr: 0x7fb5400ef008^M
[2021-12-14 08:57:52][ 2705.368472] SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1^M
[2021-12-14 08:57:52][ 2705.368482] SW_LOAD: entry[1]: addr 0x0000000000100000, size 0x000000007ff00000,  type 0x1^M
[2021-12-14 08:57:52][ 2705.368492] SW_LOAD: entry[2]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2^M
[2021-12-14 08:57:52][ 2705.368502] SW_LOAD: entry[3]: addr 0x0000000100000000, size 0x0000000080000000,  type 0x1^M
[2021-12-14 08:57:52][ 2705.368511] SW_LOAD: ovmf_entry 0xfffffff0
[2021-12-14 08:57:52][ 2705.368520] add_cpu
[2021-12-14 08:57:52][ 2705.554636] out instr on NMI port (0x61) not supported
[2021-12-14 08:57:52][ 2705.554784] out instr on NMI port (0x61) not supported
[2021-12-14 08:57:52][ 2705.609733] out instr on NMI port (0x61) not supported
[2021-12-14 08:58:15][ 2729.133465] Unhandled memory access to 0xfe001818
[2021-12-14 08:58:15][ 2729.133491] Failed to emulate instruction [[2021-12-14 08:58:15][ 2729.133499] mmio address 0xfe001818, size 4
```

I will immediately try to apply the patch to my build. Hopefully, I have enough knowledge to do that. Should I execute `vmexit clear` on the ACRN shell? Currently, I get `Error: Invalid command.` when I do that. Does this command come with the patch you mentioned.

mgcao, 2021-12-14:
yes, in ACRN shell.

christophsusen, 2021-12-14:
I applied the patch. I ran `vmexit clear` on the ACRN shell before launching the RTVM and I ran `vmexit` after launching the RTVM at the point where the start-up process becomes so slow. Here is the output:
```
Now(us) =        188168873

REASON	      CPU0	        US	      CPU1	        US	      CPU2	        US	      CPU3	        US	      CPU4	        US	      CPU5	        US	      CPU6	        US	      CPU7	        US
0x00	         0	         0	         0	         0	         0	         0	         0	         0	        37	        23	        37	        22	        37	        23	        37	        22
0x01	      1949	      1919	      2138	      2234	      1460	      2771	      1060	      1071	     12270	      9688	        20	        39	        28	        52	        36	        61
0x07	     54786	     24890	     55191	     24367	     18554	      8034	     40021	     18079	        67	        30	        78	        34	       122	        53	       185	        81
0x0a	       349	       193	       687	       369	       604	       320	       702	       374	   9433886	   3967772	       473	       249	     74483	     32322	       370	       194
0x0c	     52641	  75961664	     54016	  75033750	     18358	  74763178	     39611	  75002619	        64	   4284179	        74	   5058475	       117	   4392271	       177	   5069573
0x12	        16	     42108	         0	         0	         7	    224768	         1	         2	         0	         0	         0	         0	         0	         0	         1	        39
0x1c	         0	         0	         0	         0	         0	         0	         0	         0	        92	        64	         8	         6	         8	         5	         8	         5
0x1e	         0	         0	         8	       220	       146	      1526	        92	      2119	   4768927	   2810027	     25024	     32015	     32825	    404205	      8758	     26375
0x1f	        74	        38	        75	        39	        75	        37	        73	        37	       713	       302	        49	        23	        52	        24	        55	        28
0x20	    156714	     98555	    185496	    101262	     75158	     42928	    193897	    109207	       856	     27313	       727	      1470	       504	       718	      1046	      1849
0x28	       146	        87	        88	        51	       127	        70	        99	        56	    681303	    305027	         3	         1	         3	         2	         0	         0
0x2c	         0	         0	         0	         0	         0	         0	         0	         0	       249	       468	         0	         0	         0	         0	         0	         0
0x30	        11	       143	         5	       301	         3	        64	         1	        57	      1407	     27331	        38	       467	        74	      1194	        59	       743
0x36	         0	         0	         0	         0	         0	         0	         0	         0	         1	       301	         2	    149623	         2	    152767	         2	    155550
0x37	         0	         0	         0	         0	         0	         0	         0	         0	         2	         1	         1	         0	         1	         0	         1	         0

Now=188448312us, for detailed latency of each vmexit on each cpu:

   VMEXIT/0x00        CPU0        CPU1        CPU2        CPU3        CPU4        CPU5        CPU6        CPU7
   0us -   2us           0           0           0           0          37          37          36          37
   2us -   4us           0           0           0           0           0           0           1           0
  Max Lat(us):           0           0           0           0           0           1           2           1

   VMEXIT/0x01        CPU0        CPU1        CPU2        CPU3        CPU4        CPU5        CPU6        CPU7
   0us -   2us        1897        2138         922        1050       12261          18          24          32
   2us -   4us          23           7         517           9           7           0           2           2
   4us -   8us          23           0          22           1           0           0           0           0
   8us -  16us           5           0           0           0           1           1           1           1
  16us -  32us           1           0           0           0           0           0           0           0
  Max Lat(us):          16           2           5           4          14          14          14          14

   VMEXIT/0x07        CPU0        CPU1        CPU2        CPU3        CPU4        CPU5        CPU6        CPU7
   0us -   2us       54784       55497       18560       40021          67          78         122         185
   2us -   4us           0           1           0           0           0           0           0           0
   4us -   8us           1           0           0           0           0           0           0           0
   8us -  16us           1           0           0           0           0           0           0           0
  Max Lat(us):           8           2           1           1           0           1           0           0

   VMEXIT/0x0a        CPU0        CPU1        CPU2        CPU3        CPU4        CPU5        CPU6        CPU7
   0us -   2us         349         687         604         702     9433885         473       74483         370
   2us -   4us           0           0           0           0           1           0           0           0
[... cut ...]
8192us -  more           0           0           0           0           0           1           1           1
  Max Lat(us):           0           0           0           0           0      149249      152455      155225

   VMEXIT/0x37   VM0/vCPU0   VM0/vCPU1   VM0/vCPU2   VM0/vCPU3   VM2/vCPU0   VM2/vCPU1   VM2/vCPU2   VM2/vCPU3
   0us -   2us           0           0           0           0           2           1           1           1
  Max Lat(us):           0           0           0           0           0           0           0           0
```

mgcao, 2021-12-14:
VMEXIT/0x28   VM0/vCPU0   VM0/vCPU1   VM0/vCPU2   VM0/vCPU3   VM2/vCPU0   VM2/vCPU1   VM2/vCPU2   VM2/vCPU3
   0us -   2us         146          88         127          99      681219           0           0           0
  Max Lat(us):           1           1           1           1           1           0           0           0

This one shall not happens VM2/vCPU0, and so many times, let me check it.

christophsusen, 2021-12-14:
Happy to hear that you found something! Thanks again for helping me out!

mgcao, 2021-12-14:
@christophsusen help to confirm:

the rtvm kernel log:
  a. [    5.565435] EXT4-fs (vda2): mounted filesystem with ordered data mode. Opts: (null)
  b. [   38.293042] systemd[1]: Inserted module 'autofs4'

after the RTVM bootup, when it is stuck (go to a)
in ACRN SHELL:
  vmexit clear
when it goes to b.
  vmexit

send out the info. I'd like to confirm the data.

christophsusen, 2021-12-15:
@mgcao here you find the info that you asked for:

1. Output after executing `vmexit clear` just after the line `[    2.837401] EXT4-fs (vda2): mounted filesystem with ordered data mode. Opts: (null)`:
```
Now(us) =      57029516632

REASON	      CPU0	        US	      CPU1	        US	      CPU2	        US	      CPU3	        US	      CPU4	        US	      CPU5	        US	      CPU6	        US	      CPU7	        US

Now=57029537354us, for detailed latency of each vmexit on each cpu:

   VMEXIT/0x07        CPU0        CPU1        CPU2        CPU3        CPU4        CPU5        CPU6        CPU7
   0us -   2us           0           0           0           2           0           0           0           0
  Max Lat(us):           0           0           0           0           0           0           0           0

   VMEXIT/0x0c        CPU0        CPU1        CPU2        CPU3        CPU4        CPU5        CPU6        CPU7
8192us -  more           0           0           0           2           0           0           0           0
  Max Lat(us):           0           0           0       10048           0           0           0           0

   VMEXIT/0x1e        CPU0        CPU1        CPU2        CPU3        CPU4        CPU5        CPU6        CPU7
  Max Lat(us):           0           0           0           0           0           0           0           0

   VMEXIT/0x20        CPU0        CPU1        CPU2        CPU3        CPU4        CPU5        CPU6        CPU7
   0us -   2us           0           0           0          10           0           0           0           0
  Max Lat(us):           0           0           0           0           0           0           0           0

Now(us) =      57029662564; total vmexit per vm/vCPU (count & time):

REASON	 VM0/vCPU0	        US	 VM0/vCPU1	        US	 VM0/vCPU2	        US	 VM0/vCPU3	        US	 VM2/vCPU0	        US	 VM2/vCPU1	        US	 VM2/vCPU2	        US	 VM2/vCPU3	        US
0x07	         0	         0	         2	         0	         3	         1	        15	         7	         0	         0	         0	         0	         0	         0	         0	         0
0x0c	         0	         0	         2	    208961	         3	    149047	        15	    150724	         0	         0	         0	         0	         0	         0	         0	         0
0x1e	         0	         0	         0	         0	         0	         0	         0	         0	         0	         0	         0	         0	         1	         0	         0	         0
0x20	         0	         0	         5	         3	        13	         7	        75	        38	         0	         0	         0	         0	         0	         0	         0	         0

Now=57029756226us, for detailed latency of each vmexit on each VM/vCPU:

   VMEXIT/0x07   VM0/vCPU0   VM0/vCPU1   VM0/vCPU2   VM0/vCPU3   VM2/vCPU0   VM2/vCPU1   VM2/vCPU2   VM2/vCPU3
   0us -   2us           0           2           3          24           0           0           0           0
  Max Lat(us):           0           0           0           1           0           0           0           0

   VMEXIT/0x0c   VM0/vCPU0   VM0/vCPU1   VM0/vCPU2   VM0/vCPU3   VM2/vCPU0   VM2/vCPU1   VM2/vCPU2   VM2/vCPU3
8192us -  more           0           2           3          24           0           0           0           0
  Max Lat(us):           0      127871       67993       10048           0           0           0           0

   VMEXIT/0x1e   VM0/vCPU0   VM0/vCPU1   VM0/vCPU2   VM0/vCPU3   VM2/vCPU0   VM2/vCPU1   VM2/vCPU2   VM2/vCPU3
  Max Lat(us):           0           0           0           0           0           0           0           0

   VMEXIT/0x20   VM0/vCPU0   VM0/vCPU1   VM0/vCPU2   VM0/vCPU3   VM2/vCPU0   VM2/vCPU1   VM2/vCPU2   VM2/vCPU3
   0us -   2us           0           5          13         120           0           0           0           0
  Max Lat(us):           0           1           1           0           0           0           0           0
```

2. Output after executing `vmexit` just after the line `[   36.244839] systemd[1]: Inserted module 'autofs4'` appeared (maybe a little later since a few log lines appear faster again at that point):

```
Now(us) =      57064956624

REASON	      CPU0	        US	      CPU1	        US	      CPU2	        US	      CPU3	        US	      CPU4	        US	      CPU5	        US	      CPU6	        US	      CPU7	        US
0x01	       125	       152	       895	       854	       119	       114	        65	        56	         0	         0	         0	         0	         0	         0	         0	         0
0x07	      1144	       515	     34030	     14474	      5128	      2168	      3996	      1698	         0	         0	         0	         0	         0	         0	         0	         0
0x0a	         1	         1	         1	         0	         1	         1	         1	         0	        35	        18	         5	         3	        21	        11	        35	        19
0x0c	      1066	  35198160	     33776	  34904475	      5084	  35191024	      3978	  35391482	         0	         0	         0	         0	         0	         0	         0	         0
0x1e	         0	         0	         0	         0	         0	         0	         0	         0	         0	         0	      1527	       988	        80	    365524	         0	         0
0x1f	        37	        18	        37	        18	        36	        17	        37	        18	         0	         0	         0	         0	         0	         0	         0	         0
[... cut ...]
   4us -   8us           0           0           0           0           0           1           0           0
  Max Lat(us):           2           1           2           1           3           4           3           3

   VMEXIT/0x28   VM0/vCPU0   VM0/vCPU1   VM0/vCPU2   VM0/vCPU3   VM2/vCPU0   VM2/vCPU1   VM2/vCPU2   VM2/vCPU3
   0us -   2us           1           0           1           1           0           0           0           0
  Max Lat(us):           1           0           0           0           0           0           0           0
```

mgcao, 2021-12-15:
just check the latest data  about VMEXIT/028, there is no data on VMEXIT/0x28 VM2 CPUs, it is fine. the former data is from the initial time. So my doubt is wrong. No obvious issue in ACRN hypervisor now. Maybe we need check where the kernel goes and what task caused the long time stuck.

mgcao, 2021-12-15:
it could be a complex debugging work. I'm not sure if you're familiar with the linux kernel. 
[    5.565435] EXT4-fs (vda2): mounted filesystem with ordered data mode. Opts: (null)
[   38.293042] systemd[1]: Inserted module 'autofs4'

between them, could be something wrong caused the cpu stuck.

christophsusen, 2021-12-15:
Okay, thanks for your help! I don't think that I know enough about the Linux kernel to do that kind of debugging.

Do you have a suggestion on which guest OS to use to get an ACRN VM with RT capabilities up and running easily?

mgcao, 2021-12-15:
maybe this data: VMEXIT/0x1e   --> it cause 363129       us,  but less than 1 second, however it is not normal

mgcao, 2021-12-15:
I'm not sure if you can use ACRNTrace to catch the data, parser it, it is a port IO access; or you can do more test, to check if each time it case long time, usually it shall be <1 ms.

mgcao, 2021-12-15:
@christophsusen  you can ask @fuzhongl  if could offer a virtio-img for RTVM to test again.

christophsusen, 2021-12-15:
@mgcao, @fuzhongl, that would be really helpful!

Unfortunately, I don't have the knowledge yet to do further debugging by myself.

Thanks for your help!

fuzhongl, 2021-12-15:
@christophsusen 
We can't provide any image for policy concern.
But, we can try to re-produce this issue on our side if you  share the image of rtvm rootfs to us.
Thanks!

BR.
Fuzhong

gvancuts, 2021-12-15:
How did you build your RT kernel by the way (what config are you using)? I see you also use an initrd with it, have you tried to have all drivers compiled in instead and get rid of that initrd?

christophsusen, 2021-12-15:
@fuzhongl, thanks for the offer! I will try to make a clean user VM image later today or tomorrow and send you a link to download it afterwards.

christophsusen, 2021-12-15:
Hi @gvancuts, as I started to experiment with ACRN, I first tried to configure a non-RTVM with Ubuntu 20.04. After I got this to work, I tried to use this VM image as a starting point for the RTVM and compiled the preempt RT kernel inside the Ubuntu VM. Here are all the commands that I executed to do this:

```
wget https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/5.10/older/patch-5.10.52-rt47.patch.gz
wget https://mirrors.edge.kernel.org/pub/linux/kernel/v5.x/linux-5.10.52.tar.gz

tar xzvf linux-5.10.52.tar.gz
gunzip patch-5.10.52-rt47.patch.gz
cd linux-5.10.52
patch -p1 < ../patch-5.10.52-rt47.patch

# Use kernel config that comes with Ubuntu as basis.
cp /boot/config-$(uname -r) .config
make olddefconfig

# Select fully preemtible kernel from the menu, leave everything else as is.
make menuconfig

scripts/config --disable SYSTEM_TRUSTED_KEYS
scripts/config --disable SYSTEM_REVOCATION_KEYS

make -j $(nproc) deb-pkg

for pkg in ./*.deb; do sudo dpkg -i $pkg; done
sudo reboot
```

The only thing I changed in the user VM from that point are the kernel command line parameters. I think, the last set I used is the following: `ro rootwait nohpet console=hvc0 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll irqaffinity=0`

Regarding your second question: I have not tried to get rid of initrd. If that could potentially solve the issue, I have to figure out how to do this first. I actually never compiled a linux kernel except for this RT kernel for the VM. :sweat_smile:

gvancuts, 2021-12-15:
Thanks @christophsusen ! @mgcao , do we have a kernel config we could recommend? Should we have @christophsusen to compile and try use https://github.com/projectacrn/acrn-kernel/tree/5.10/preempt-rt isntead (not sure whether we may have other ACRN patches in there)?

You do not need an initrd if all the essential drivers are built into the kernel, so if your kernel config is right, you can avoid it.

mgcao, 2021-12-16:
@gvancuts yes there is a config file for RT kernel, but I have not built the RT kernel.  could @fuzhongl check that?

christophsusen, 2021-12-16:
@gvancuts, @mgcao, I can't find a config file in https://github.com/projectacrn/acrn-kernel/tree/5.10/preempt-rt. Do I have to generate such a config file by myself from scratch or can you recommend something to get started with?

Thanks a lot for taking so much time to help me out, much appreciated!

gvancuts, 2021-12-16:
> @gvancuts, @mgcao, I can't find a config file in https://github.com/projectacrn/acrn-kernel/tree/5.10/preempt-rt. Do I have to generate such a config file by myself from scratch or can you recommend something to get started with?

We have one for the 4.19 kernel [here](https://github.com/projectacrn/acrn-kernel/blob/4.19/preempt-rt/x86-64_defconfig). But I have not played with this in some time and so I do not know if you can use it with the 5.10 kernel. @fuzhongl , can you share the kernel config file we use to test realtime performace?

fuzhongl, 2021-12-16:
Please try it: [rtvm_config.gz](https://github.com/projectacrn/acrn-hypervisor/files/7725547/rtvm_config.gz)

christophsusen, 2021-12-16:
Thanks, @fuzhongl!

I compiled the kernel with the config file you provided and tried to use it with the non-RTVM and with the RTVM launch script. The former one works fine, as before. Unfortunately, when using the launch script for the RTVM, the starting process gets stuck again.

Here is the output that I get on the ACRN console after executing `vm_console 2`:
```
Loading Linux 5.10.78-acrn-kernel-sos+ ...
6az at.1Orottlose o=po:AVE f1: 'x
0-0[f f0affefBIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
BIOS-e820: [mem 0x00000000ffe00000-0x00000000ffe1ffff] reserved
BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
printk: debug: ignoring loglevel setting.
process: using polling idle threads
NX (Execute Disable) protection: active
efi: EFI v2.70 by ARCN
efi: ACPI=0x7f9fa000 ACPI 2.0=0x7f9fa014 MEMATTR=0x7f55a218
DMI not present or invalid.
Hypervisor detected: ACRN
tsc: Detected 3000.000 MHz processor
e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
e820: remove [mem 0x000a0000-0x000fffff] usable
last_pfn = 0x180000 max_arch_pfn = 0x400000000
MTRR default type: uncachable
MTRR fixed ranges enabled:
  00000-FFFFF write-back
MTRR variable ranges enabled:
x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
CPU MTRRs all blank - virtualized system.
last_pfn = 0x7ffe0 max_arch_pfn = 0x400000000
found SMP MP-table at [mem 0x000f0000-0x000f000f]
check: Scanning 1 areas for low memory corruption
Using GB pages for direct mapping
printk: log_buf_len: 16777216 bytes
printk: early log buf free: 127840(97%)
Secure boot disabled
ACPI: Early table checksum verification disabled
ACPI: RSDP 0x000000007F9FA014 000024 (v02 DM    )
ACPI: XSDT 0x000000007F9F90E8 000044 (v01 DM     DMFACP   00000001      01000013)
ACPI: FACP 0x000000007F9F7000 00010C (v05 DM     DMFACP   00000001 INTL 20210105)
ACPI: DSDT 0x000000007F9F4000 000336 (v02 DM     DMDSDT   00000001 INTL 20210105)
ACPI: FACS 0x000000007F9FE000 000040
ACPI: APIC 0x000000007F9F8000 00005E (v01 DM     DMMADT   00000001 INTL 20210105)
ACPI: HPET 0x000000007F9F6000 000038 (v01 DM     DMHPET   00000001 INTL 20210105)
ACPI: MCFG 0x000000007F9F5000 00003C (v01 DM     DMMCFG   00000001 INTL 20210105)
ACPI: Reserving FACP table memory at [mem 0x7f9f7000-0x7f9f710b]
ACPI: Reserving DSDT table memory at [mem 0x7f9f4000-0x7f9f4335]
ACPI: Reserving FACS table memory at [mem 0x7f9fe000-0x7f9fe03f]
ACPI: Reserving APIC table memory at [mem 0x7f9f8000-0x7f9f805d]
ACPI: Reserving HPET table memory at [mem 0x7f9f6000-0x7f9f6037]
ACPI: Reserving MCFG table memory at [mem 0x7f9f5000-0x7f9f503b]
ACPI: Local APIC address 0xfee00000
Zone ranges:
  DMA      [mem 0x0000000000001000-0x0000000000ffffff]
  DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
  Normal   [mem 0x0000000100000000-0x000000017fffffff]
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000000001000-0x000000000009ffff]
  node   0: [mem 0x0000000000100000-0x000000007e7d1fff]
  node   0: [mem 0x000000007e7d3000-0x000000007e7f4fff]
  node   0: [mem 0x000000007e91b000-0x000000007f99afff]
  node   0: [mem 0x000000007f9ef000-0x000000007f9f3fff]
  node   0: [mem 0x000000007f9ff000-0x000000007ffdffff]
  node   0: [mem 0x0000000100000000-0x000000017fffffff]
Initmem setup node 0 [mem 0x0000000000001000-0x000000017fffffff]
On node 0 totalpages: 1048057
  DMA zone: 64 pages used for memmap
  DMA zone: 2039 pages reserved
  DMA zone: 3999 pages, LIFO batch:0
  DMA32 zone: 8122 pages used for memmap
[... 265 lines cut ...]
virtio-pci 0000:00:09.0: PCI INT A: not connected
virtio-pci 0000:00:09.0: virtio_pci: leaving for legacy driver
Serial: 8250/16550 driver, 16 ports, IRQ sharing disabled
[121209210us][cpu=6][vm2:vcpu2][sev=3][seq=334]:vioapic: WARNING: invalid vpic wire mode change
[121345502us][cpu=6][vm2:vcpu2][sev=3][seq=335]:vioapic: WARNING: invalid vpic wire mode change
serial8250: ttyS0 at I/O 0x3f8 (irq = 0, base_baud = 115200) is a 16550A
[121480207us][cpu=6][vm2:vcpu2][sev=3][seq=336]:vioapic: WARNING: invalid vpic wire mode change
serial8250: ttyS1 at I/O 0x2f8 (irq = 0, base_baud = 115200) is a 16550A
[121643910us][cpu=3][vm0:vcpu3][sev=3][seq=337]:ret=-1 hypercall=0x80000023 failed in vmcall_vmexit_handler

[121654365us][cpu=3][vm0:vcpu3][sev=3][seq=338]:ret=-1 hypercall=0x80000023 failed in vmcall_vmexit_handler

[121664877us][cpu=3][vm0:vcpu3][sev=3][seq=339]:ret=-1 hypercall=0x80000023 failed in vmcall_vmexit_handler

[121681385us][cpu=1][vm0:vcpu1][sev=3][seq=340]:ret=-1 hypercall=0x80000023 failed in vmcall_vmexit_handler

hpet_acpi_add: no address or irqs in _CRS
Linux agpgart interface v0.103
brd: module loaded
loop: module loaded
virtio_blk virtio0: [vda] 209715200 512-byte logical blocks (107 GB/100 GiB)
vda: detected capacity change from 0 to 107374182400
clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x2b3e459bf4c, max_idle_ns: 440795289890 ns
clocksource: Switched to clocksource tsc
```

christophsusen, 2021-12-17:
Just in case this might help to narrow down the issue: I tried to launch the Ubuntu user VM with Xenomai kernel 4.19.59 as described on [this](https://projectacrn.github.io/latest/tutorials/using_xenomai_as_user_vm.html). page in the ACRN documentation. Using this kernel, the RTVM boots without any problems. So it seems like the problem is somewhere in the kernel configuration.

Do you have advice for me on how to configure a preempt-RT kernel maybe based on the working configuration for the Xenomai kernel?

christophsusen, 2022-02-13:
After a long time, I was able to work on this issue again and finally got everything to work. I pursued the following steps. Just in case, this might help someone else.

- Clone 5.10/preempt-rt branch from https://github.com/projectacrn/acrn-kernel.git
- Use kernel configuration file provided by @fuzhongl above (https://github.com/projectacrn/acrn-hypervisor/issues/6950#issuecomment-995537698)
- Enable polling mode for VIRTIO devices using make menuconfig and build the kernel

Thanks a lot for your help, @fuzhongl, @gvancuts and @mgcao

closed 2022-02-13 by christophsusen