projectacrn/acrn-hypervisor #6981: The event ring segment size should be at the xHCI spec define value rang

liudlong, 2021-12-16:
Environment:

 

HV : v2.7-rc5-395174f9d

kernel: 5.10.65-acrn-kernel-sos

HW: ADL-S-CRB

iso: \\10.239.48.225\acrn\ACRN_Validation\acrn-hypercube\v2.7\new-iso

launch script: \\10.239.48.225\acrn\ACRN_Validation\acrn-hypercube\v2.7\launch.sh

 

reproduce steps:

apply patch and build acrn
replace build image and copy iso and launch script to target Board
run hypercube_PCI_USB.iso with script launch.sh
Expected Result

no error happened during PCI_USB hypercube testing

Actual result:

ERROR: FPE on unknown address 0x55849f31ab45

logs:

hw/pci/xhci.c:3342:19: runtime error: index 157 out of bounds for type 'pci_xhci_dev_ep [32]'
hw/pci/xhci.c:1866:43: runtime error: division by zero
ASAN:DEADLYSIGNAL
=================================================================
==572==ERROR:  FPE on unknown address 0x55849f31ab45 (pc 0x55849f31ab45 bp 0x7f829e9199a0 sp 0x7f829e919940 T16)
#0 0x55849f31ab44 in pci_xhci_insert_event hw/pci/xhci.c:1866
#1 0x55849f33e684 in pci_xhci_complete_commands hw/pci/xhci.c:2771
#2 0x55849f33edcb in pci_xhci_dbregs_write hw/pci/xhci.c:3419
#3 0x55849f33f1e6 in pci_xhci_write hw/pci/xhci.c:3657
#4 0x55849f345dce in pci_emul_mem_handler hw/pci/core.c:527
#5 0x55849f40e529 in mem_write core/mem.c:147
#6 0x55849f415880 in emulate_mem core/mem.c:186
#7 0x55849f41fc8d in vmexit_mmio_emul core/main.c:331
#8 0x55849f41f13b in handle_vmexit core/main.c:409
#9 0x55849f41f561 in vm_loop core/main.c:709
#10 0x55849f420349 in start_thread core/main.c:251
#11 0x7f8469f5aea3 (/lib/libpthread.so.0+0x8ea3)
#12 0x7f8468d23dfe in clone (/lib/libc.so.6+0xf8dfe)

liudlong, 2021-12-16:
[External_System_ID] ACRN-7546

NanlinXie, 2022-01-15:
Closed this issue after verification.

closed 2022-01-15 by NanlinXie