projectacrn/acrn-hypervisor #6983: How to activate Interrupt from prelaunched Linux VM via ivshmem device

hidrol, 2021-12-16:
I'm using two prelaunched vms: rt linux and zephyr with ivshmem communication. 

Now I'd like to be able to activate Interrupts from the linux vm to the zephyr vm via ivshmem device.
I was able to open /sys/class/uio/uioX/device/resource1 and map the memory with mmap() but I don't know
how to activate the Doorbell Register IVSHMEM_DOORBELL_REG.
Also I'm not shure where I have to write the VM ID I want to interrupt.

I read under https://projectacrn.github.io/latest/developer-guides/hld/ivshmem-hld.html and 
https://www.kernel.org/doc/html/v4.12/driver-api/uio-howto.html but couldn't find enough information.

hidrol, 2021-12-16:
https://github.com/qemu/qemu/blob/master/docs/specs/ivshmem-spec.txt
includes the information I needed:
BAR 0 contains the following registers:

```
    Offset  Size  Access      On reset  Function
        0     4   read/write        0   Interrupt Mask
                                        bit 0: peer interrupt (rev 0)
                                               reserved       (rev 1)
                                        bit 1..31: reserved
        4     4   read/write        0   Interrupt Status
                                        bit 0: peer interrupt (rev 0)
                                               reserved       (rev 1)
                                        bit 1..31: reserved
        8     4   read-only   0 or ID   IVPosition
       12     4   write-only      N/A   Doorbell
                                        bit 0..15: vector
                                        bit 16..31: peer ID
       16   240   none            N/A   reserve
```d

dbkinder, 2021-12-16:
@NanlinXie This looks like something we should add to the ACRN documentation. Can you assign an ACRN developer to help update our IVSHMEM documentation with information from this GitHub issue.

NanlinXie, 2022-01-15:
@yonghuah , any comments for this request?

closed 2022-07-05 by fuzhongl