projectacrn/acrn-hypervisor #6990: Add GRUB version pre-requisite and instructions

dbkinder, 2021-12-17:
There's a known issue with GRUB that could require developers to manually upgrade their GRUB version.  This was casually documented in the [v2.4 Getting Started Guide for ACRN Industry Scenario With Ubuntu Service VM](https://projectacrn.github.io/2.4/getting-started/rt_industry_ubuntu.html#prerequisites).  Alas, this mention was [removed in v2.5](https://github.com/projectacrn/acrn-hypervisor/pull/6215/files) and the whole document disappeared with v2.6.

ACRN does not support GRUB putting a module at address 0 or under 1MB when the module size is less than 1MB (enforced by code in ``acrn-hypervisor/hypervisor/boot/guest/vboot_info.c``.  The out-of-the-box installation of Ubuntu 18.04 or even 20.04 does not install a GRUB version that has been [patched to fix this issue](http://git.savannah.gnu.org/cgit/grub.git/commit/?id=0f3f5b7c13fa9b677a64cf11f20eca0f850a2b20): *multiboot2: Set min address for mbi allocation to 0x1000*

Depending on target platform memory layout and loaded module size, a developer could hit an error while loading, saying:
```
[21602370us][cpu=3][(null)][sev=3][seq=27]:Unsupported module: start at HPA 0, size 0x7e998
```
--------
- Document this issue and point out that the latest version of GRUB [available from the GNU site (2.06)](https://www.gnu.org/software/grub/manual/grub/grub.html#Obtaining-and-Building-GRUB) does have this patch and should be used if this "Unsupported module start" error is encountered. 
- Update the error message in ``vboot_info.c`` to mention updating GRUB to resolve this reported problem.

dbkinder, 2021-12-21:
Padding your (Zephyr) ``.elf`` image to 1MB (since we don’t support pre-launched VM OS image less than 1MB) can be done with a command like this:
```
dd if=zephyr.elf ibs=1M count=1 of=zephyr_new.elf conv=sync
```
The padded version of ``zephyr.elf`` would be ``zephyr_new.elf``

gvancuts, 2021-12-21:
> Padding your (Zephyr) `.elf` image to 1MB (since we don’t support pre-launched VM OS image less than 1MB) can be done with a command like this:
> 
> ```
> dd if=zephyr.elf ibs=1M count=1 of=zephyr_new.elf conv=sync
> ```
> 
> The padded version of `zephyr.elf` would be `zephyr_new.elf`

I would suggest to use a different name, i.e. `zephyr_1MB.elf`.

We also need to make sure we do not run this command if the `zephyr.elf` file is greater than 1MB because otherwise it will actually truncate it.