projectacrn/acrn-hypervisor #6994: Failed to start acrnd service in Service VM

wuxyintel, 2021-12-20:
Currently, if user starts acrnd service, acrnd service will fail to find the folder "/sys/kernel/gvt" in Service VM.
Root cause is GVT-g is not supported in current ACRN, the folder "/sys/kernel/gvt" will not be created in Service VM.

gvancuts, 2021-12-20:
Out of curiosity, is the `After=weston.service` line not a problem?

What is the OS running in the Service VM here?

wuxyintel, 2021-12-21:
Hi Geoffory, weston is the reference implementation of a Wayland server. A Wayland server is a display server, a window manager, and a compositor all in one. Weston service is optional for ACRN, so I will remove this configuration in acrnd service. I verified it on Ubuntu Service VM.

gvancuts, 2021-12-21:
> Hi Geoffory, weston is the reference implementation of a Wayland server. A Wayland server is a display server, a window manager, and a compositor all in one. Weston service is optional for ACRN, so I will remove this configuration in acrnd service. I verified it on Ubuntu Service VM. Thanks!

Thanks!

wuxyintel, 2022-01-04:
[External_System_ID] ACRN-7575

closed 2022-03-17 by fuzhongl