projectacrn/acrn-hypervisor #7003: Send only one character on legacy uart between windows vm to other vm

axiomtek-yichen, 2022-01-03:
**Describe the bug**
Can not send string from windows vm to ohter vm
**Platform**
Coffee lake i7
**Codebase**
ACRN 2.5
**Scenario**
industry
**To Reproduce**
1. Go to Windows VM open putty on COM2 and SOS VM open minicom /dev/ttyS1
2. Can send character and string through SOS to Windows VM
3. Windows VM can only send a character to SOS VM not only putty or write comport program by myself

**Expected behavior**
Windows should send not only character but string to other vm

xiaoxiax, 2022-01-20:
Can’t reproduced the issue on CFL-K700-i7 with V2.5 release.

Reproduce steps:
1.	Rebuild acrn.bin 
change scenario xml(industry.xml):
<vm id="0">
.....
<legacy_vuart id="1">
      <type>VUART_LEGACY_PIO</type>
      <base>SERVICE_VM_COM2_BASE</base>
      <irq>SERVICE_VM_COM2_IRQ</irq>
      <target_vm_id>1</target_vm_id>
      <target_uart_id>1</target_uart_id>
 </legacy_vuart>
---------------------------------------
<vm id="1">
.......
<legacy_vuart id="1">
   <type>VUART_LEGACY_PIO</type>
      <base>COM2_BASE</base>
      <irq>COM2_IRQ</irq>
      <target_vm_id>0</target_vm_id>
      <target_uart_id>1</target_uart_id>
  </legacy_vuart>
---------------------------------------
<vm id="2">
.......
<legacy_vuart id="1">
   <type>VUART_LEGACY_PIO</type>
      <base>INVALID_COM_BASE</base>
      <irq>COM2_IRQ</irq>
      <target_vm_id>0</target_vm_id>
      <target_uart_id>1</target_uart_id>
  </legacy_vuart>
2. Launch waag and Install putty from \\10.239.48.225\acrn\ACRN_Validation\IO_Mediators\putty.exe
3. Open putty and set connection type to "Serial" , Serial line to "COM2", Speed to "115200"
4. echo "test" > /dev/ttyS2 on SOS and see return value in putty on waag
5. Cat  /dev/ttyS2 on SOS and  input characters into putty on waag

yichen-ax, 2022-01-20:
Left monitor is SOS right monitor is windows vm

The problem is as below
https://youtu.be/L1N_8RDYlQ0
https://youtu.be/5G2ttnawLr0

I discovered this problem is when I set S5 windows has to send reboot to sos
but windows can only sent one character can not set string so I could not set s5 in windows vm

gvancuts, 2022-02-15:
Is this issue still valid?

yichen-ax, 2022-02-16:
> 

Yes, still don't know how to solve it.

gvancuts, 2022-02-16:
Can you share your scenario file, ACRN version (still 2.5?) and serial port configuration (inc. config in Putty)?

fuzhongl, 2022-07-05:
@yichen-ax Please refer following doc for S5: 
https://projectacrn.github.io/latest/tutorials/enable_s5.html

closed 2022-07-05 by fuzhongl