projectacrn/acrn-hypervisor #7015: board inspector Base Board Information missing on Vecow SPC-7100

dbkinder, 2022-01-11:
**Describe the bug**
On the Vecow SPC-7100 system, the Base Board Info reported by the board inspector is missing (all values are "Default string")

**Platform**
[Vecow SPC 7100 (Tiger Lake - U)](https://www.vecow.com/dispPageBox/vecow/VecowCT.aspx?ddsPageID=PRODUCTDTL_EN&dbid=4616165062)

**Codebase**
master

**More**
After (successfully) running the configurator, I notice the generated ``vecow.xml`` file starts with this:
```
<acrn-config board="vecow">
  <BIOS_INFO>
      BIOS Information
      Vendor: American Megatrends International, LLC.
      Version: E5000XXU3F00105-BPCIe
      Release Date: 08/25/2021
      BIOS Revision: 5.19
      </BIOS_INFO>
  <BASE_BOARD_INFO>
      Base Board Information
      Manufacturer: Default string
      Product Name: Default string
      Version: Default string
      </BASE_BOARD_INFO>
```

The configurator gets this base board info by running a ``dmidecode -t 2`` command.  If I manually run this, I see:

```
:~$ sudo dmidecode -t 2
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: Default string
        Product Name: Default string
        Version: Default string
        Serial Number: Default string
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0
```

It would appear that ``dmidecode`` isn't working properly to retrieve the expected data from this platform, causing the configurator to report incorrect information.

gvancuts, 2022-01-12:
Have you tried to (manually) update `dmidecode` to 3.3? I noticed this message in the output:
```
SMBIOS 3.3.0 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.
```

dbkinder, 2022-01-12:
Same "Default string" output when using dmidecode version 3.3 (built from GitHub source).

fuzhongl, 2022-07-05:
[image]
Fixed with this BIOS version.

closed 2022-07-05 by fuzhongl