projectacrn/acrn-hypervisor #7022: ACRN debian package not complete when source is not cloned to standard folder

shiqingg, 2022-01-13:
When cloning the ACRN source code to a folder, which is not named "acrn-hypervisor", the debian package created by the "make" call does not contain all needed files.
This is most likely due to the fact, that the file ".deb.conf" contains hardcoded folder names which contain the folder "acrn-hypervisor". Nevertheless, the build does not fails, which it should...

shiqingg, 2022-01-13:
[External_System_ID] ACRN-7584

NanlinXie, 2022-01-15:
@shiqingg , could you confirm whether this issue is resolved by merged patch?

shiqingg, 2022-01-17:
> @shiqingg , could you confirm whether this issue is resolved by merged patch?

@NanlinXie Yes, it can be solved by this patch. PR#7016 for master branch. PR#7023 for release_2.7 branch.

NanlinXie, 2022-01-18:
Closed this issue with verified result.