projectacrn/acrn-hypervisor #7024: Can Config Tool's passthru device add like virtio devices?

yichen-ax, 2022-01-13:
Each kind of pcie device can set one passthru device only.
I use 4-port GigE PCI Express Card I can only set one ethernet passthru device on vm.
Will config tool let passthru device addable in the future?
Thanks

NanlinXie, 2022-01-15:
Do you mean you want to passthrough multiple device with the same type?
@yuchuyang is ongoing refining config tool to support this, and she could explain more about implementation..

yichen-ax, 2022-01-17:
Hi NanlinXie
Thanks for your reply.
How can I know the roadmap of config tool about support multiple passthrough.
Should I ask yuchuyang directly?

gvancuts, 2022-01-17:
> Hi NanlinXie Thanks for your reply. How can I know the roadmap of config tool about support multiple passthrough. Should I ask yuchuyang directly?

Let's make sure the response gets posted here so if others are interested they will know, or can find out later.

yuchuyang, 2022-01-20:
Hi  @yichen-ax,
I have some general questions for your scenario:
1. Did you try to launch a post-launched VM of shared or hybrid mode?
2. Did you try to pass through multiple ethernet cards for a post-launched VM?
3. Did you generate a launch_xxxx.sh for a post-launched VM?

yichen-ax, 2022-01-20:
> Hi @yichen-ax, I have some general questions for your scenario:
> 
> 1. Did you try to launch a post-launched VM of shared or hybrid mode?
> 2. Did you try to pass through multiple ethernet cards for a post-launched VM?
> 3. Did you generate a launch_xxxx.sh for a post-launched VM using \misc\config_tools\launch_config\launch_cfg_gen.py?

1.VM shared
2.yes
3.yes

What I mean passthorough device launch.sh is as below
#!/bin/bash
# board: SBC87525, scenario: INDUSTRY_AX, uos: WINDOWS
# pci devices for passthru
declare -A passthru_vpid
declare -A passthru_bdf

passthru_vpid=(
["usb_xdci"]="8086 a36d"
["pci_device1"]="8086 1539"
["pci_device2"]="8086 1539"
["pci_device3"]="8086 1539"
["pci_device4"]="8086 1539"
["gpu"]="8086 3e92"
)
passthru_bdf=(
["usb_xdci"]="0000:00:14.0"
["pci_device1"]="0000:03:00.0"
["pci_device2"]="0000:04:00.0"
["pci_device3"]="0000:05:00.0"
["pci_device4"]="0000:06:00.0"
["gpu"]="0000:00:02.0"
)

function tap_net() {
# create a unique tap device for each VM
tap=$1
tap_exist=$(ip a | grep "$tap" | awk '{print $1}')
if [ "$tap_exist"x != "x" ]; then
  echo "tap device existed, reuse $tap"
else
  ip tuntap add dev $tap mode tap
fi

# if acrn-br0 exists, add VM's unique tap device under it
br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
  echo "acrn-br0 bridge aleady exists, adding new tap device to it..."
  ip link set "$tap" master acrn-br0
  ip link set dev "$tap" down
  ip link set dev "$tap" up
fi
}

function launch_windows()
{
#vm-name used to generate uos-mac address
mac=$(cat /sys/class/net/e*/address)
vm_name=post_vm_id$1
mac_seed=${mac:0:17}-${vm_name}

tap_net tap_WaaG
#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep -w "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

echo ${passthru_vpid["gpu"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["gpu"]} > /sys/bus/pci/devices/${passthru_bdf["gpu"]}/driver/unbind
echo ${passthru_bdf["gpu"]} > /sys/bus/pci/drivers/pci-stub/bind
modprobe pci_stub
# Passthrough USB_XDCI
echo ${passthru_vpid["usb_xdci"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["usb_xdci"]} > /sys/bus/pci/devices/${passthru_bdf["usb_xdci"]}/driver/unbind
echo ${passthru_bdf["usb_xdci"]} > /sys/bus/pci/drivers/pci-stub/bind

# Passthrough PCI_DEVICE1
echo ${passthru_vpid["pci_device1"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["pci_device1"]} > /sys/bus/pci/devices/${passthru_bdf["pci_device1"]}/driver/unbind
echo ${passthru_bdf["pci_device1"]} > /sys/bus/pci/drivers/pci-stub/bind

# Passthrough PCI_DEVICE2
echo ${passthru_vpid["pci_device2"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["pci_device2"]} > /sys/bus/pci/devices/${passthru_bdf["pci_device2"]}/driver/unbind
echo ${passthru_bdf["pci_device2"]} > /sys/bus/pci/drivers/pci-stub/bind

# Passthrough PCI_DEVICE3
echo ${passthru_vpid["pci_device3"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["pci_device3"]} > /sys/bus/pci/devices/${passthru_bdf["pci_device3"]}/driver/unbind
echo ${passthru_bdf["pci_device3"]} > /sys/bus/pci/drivers/pci-stub/bind

# Passthrough PCI_DEVICE4
echo ${passthru_vpid["pci_device4"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["pci_device4"]} > /sys/bus/pci/devices/${passthru_bdf["pci_device4"]}/driver/unbind
echo ${passthru_bdf["pci_device4"]} > /sys/bus/pci/drivers/pci-stub/bind

mem_size=4096M
#interrupt storm monitor for pass-through devices, params order:
#threshold/s,probe-period(s),intr-inject-delay-time(ms),delay-duration(ms)
intr_storm_monitor="--intr_monitor 10000,10,1,100"

#logger_setting, format: logger_name,level; like following
logger_setting="--logger_setting console,level=4;kmsg,level=3;disk,level=5"

acrn-dm -A -m $mem_size -s 0:0,hostbridge -U d2795438-25d6-11e8-864e-cb7a18b34643 \
   --windows \
   --pm_notify_channel uart,allow_trigger_s5 --pm_by_vuart tty,/dev/ttyS1 \
   $logger_setting \
   -s 10,virtio-blk,/home/acrn/win-old.img \
   -s 11,virtio-net,tap_WaaG \
   -s 2,passthru,0/2/0,gpu  \
   --ovmf /usr/share/acrn/bios/OVMF.fd \
   --cpu_affinity 3,4,5 \
   $intr_storm_monitor \
   -s 31:0,lpc \
   -l com1,stdio \
   -s 4,passthru,00/14/0,d3hot_reset \
   -s 5,passthru,03/00/0 \
   -s 6,passthru,04/00/0 \
   -s 7,passthru,05/00/0 \
   -s 8,passthru,06/00/0 \
   $vm_name
}
launch_windows 1

yuchuyang, 2022-01-21:
Hi @yichen-ax, thank you for reply. So, Yes, there is a restriction in configuration tool that the generated launch.sh cannot pass through the same category device more than one(In your case, it's ethernet). We are currently working in progress to refine the framework in backend and this may be resolve in next release.
In short term, you may pass through more than one Ethernet cards by manual editing. Add a line similar to "-s #,passthru,xx/xx/x" may work, # is the virtual slot, add one that is not being used, you can use #=9 based on your post. xx/xx/x is the device physical bdf from native environment.

closed 2022-06-23 by fuzhongl