projectacrn/acrn-hypervisor #7052: {"launch config error":"Replacement index 1 out of range for positional args tuple"}

coveritytest, 2022-01-21:
**Describe the bug**
The error {"launch config error":"Replacement index 1 out of range for positional args tuple"} pops up when clicking on "Generate launch script" in http://127.0.0.1:5001/launch/shared_launch_6user_vm

NanlinXie, 2022-01-24:
@coveritytest Could you explain more details about your issue, below template for your reference.

**Describe the bug**
A clear and concise description of what the bug is.

**Platform**
Describe the platform information, 

**Codebase**
Describe which codebase is used.

**Scenario**
Describe the scenario: Hybrid, Industry or logical_partition

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

fuzhongl, 2022-03-22:
@coveritytest Could you please help to have a try with the latest code?
Thanks!

fuzhongl, 2022-07-26:
Closed since no feedback long time.

closed 2022-07-26 by fuzhongl