projectacrn/acrn-hypervisor #7069: Error building ACRN for a NUC11PAHi5 board

gvancuts, 2022-01-27:
Steps followed:
- Installed Ubuntu 20.04 on NUC11PAHi5
- Installed latest updates
- Built the `board_inspector` utility and installed that `.deb` package on the target
- Ran it (produced some warnings):
```
$ sudo board_inspector.py my-NUC11PAHi5
[sudo] password for acrn: 
Warning: The Cx data for ACRN relies on acpi_idle driver but it is not found, please add idle=nomwait in kernel cmdline to fall back to acpi_idle driver
Warning: The Px data for ACRN relies on acpi-cpufreq driver but it is not found, please add intel_pstate=disable in kernel cmdline to fall back to acpi-cpufreq driver
WARNING:root:Parse ACPI DSDT/SSDT failed: [Errno 1] Operation not permitted: '/dev/mem'
WARNING:root:Will not extract information from ACPI DSDT/SSDT
my-NUC11PAHi5.xml saved successfully!
```

- Transfered the board XML file back to my dev machine
- Started the ACRN configurator
- Imported the board XML
- Loaded the default `shared` scenario XML
- Attempted to build ACRN with these:
```
$ make BOARD=../my-NUC11PAHi5/my-NUC11PAHi5.xml SCENARIO=../my-NUC11PAHi5/shared.xml 
make[1]: Entering directory '/home/gvancuts/work/acrn-hypervisor/hypervisor'
Board XML is fetched from /home/gvancuts/work/my-NUC11PAHi5/my-NUC11PAHi5.xml
Scneario XML is configuration fetched from /home/gvancuts/work/my-NUC11PAHi5/shared.xml
Traceback (most recent call last):
  File "/home/gvancuts/work/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 41, in <module>
    main(args)
  File "/home/gvancuts/work/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 31, in main
    module.fn(board_etree, scenario_etree, allocation_etree)
  File "/home/gvancuts/work/acrn-hypervisor/misc/config_tools/static_allocators/pio.py", line 66, in fn
    raise lib.error.ResourceError(f"{hv_debug_console} is not in the native environment! The ttyS available are: {native_ttys.keys()}")
lib.error.ResourceError:  is not in the native environment! The ttyS available are: dict_keys([])
make[1]: *** No rule to make target '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/allocation.xml', needed by '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/unified.xml'.  Stop.
make[1]: Leaving directory '/home/gvancuts/work/acrn-hypervisor/hypervisor'
make: *** [Makefile:116: hvdefconfig] Error 2
```

ACRN version:
```
$ git describe 
acrn-2022w04.5-180000p-9-g8b7a041b5
```

gvancuts, 2022-01-27:
Here is the board XML file (see attached, changed extension to `.txt`)
[my-NUC11PAHi5.txt](https://github.com/projectacrn/acrn-hypervisor/files/7950501/my-NUC11PAHi5.txt)
)

I have also tried to set the `RELEASE` flag to 'y' so as to not need a debug console but the build error remains.

gvancuts, 2022-01-27:
I had forgotten to reboot the platform before generating the board XML, here is the latest file (and `board_inspector` output):
```
acrn@acrn-tgl-nuc:~$ cat /proc/cmdline 
BOOT_IMAGE=/boot/vmlinuz-5.13.0-27-generic root=UUID=50131b44-fc2b-40c0-9782-3067f6e42c61 ro quiet splash idle=nomwait iomem=relaxed intel_idle.max_cstate=0 intel_pstate=disable vt.handoff=7
acrn@acrn-tgl-nuc:~$ sudo board_inspector.py  my-NUC11PAHi5
[sudo] password for acrn: 
WARNING:root:Parse ACPI DSDT/SSDT failed: [Errno 1] Operation not permitted: '/dev/mem'
WARNING:root:Will not extract information from ACPI DSDT/SSDT
my-NUC11PAHi5.xml saved successfully!
[my-NUC11PAHi5.txt](https://github.com/projectacrn/acrn-hypervisor/files/7950565/my-NUC11PAHi5.txt)

```

fuzhongl, 2022-04-19:
[External_System_ID] ACRN-8811

fuzhongl, 2022-04-19:
@gvancuts Using master branch, the release version build successfully on my side with you shared board xml:
ACRN Configuration Summary:
Build directory: /home/acrn/acrn-hypervisor/build/hypervisor
```
This build directory is configured with the settings below.
- BOARD = my-NUC11PAHi5
- SCENARIO = shared
- RELEASE = y
```
[shared.txt](https://github.com/projectacrn/acrn-hypervisor/files/8511774/shared.txt)

gvancuts, 2022-04-26:
Thanks @fuzhongl , I do not have the board available to me anymore. @leonmatt, could you validate that it now works on master?

leonmatt, 2022-04-26:
@gvancuts After doing a clean clone of the repo, using my-NUC11PAHi5 and shared.xml config files, both of which are attached to this issue, I get the following error:

Board XML is fetched from /home/mleon/Desktop/development/configs_modified/temporary/my-NUC11PAHi5.txt
Scenario XML is being fetched from /home/mleon/Desktop/development/configs_modified/temporary/shared.txt
Traceback (most recent call last):
  File "/home/mleon/Desktop/development/acrn/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 41, in <module>
    main(args)
  File "/home/mleon/Desktop/development/acrn/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 31, in main
    module.fn(board_etree, scenario_etree, allocation_etree)
  File "/home/mleon/Desktop/development/acrn/acrn-hypervisor/misc/config_tools/static_allocators/clos.py", line 149, in fn
    creat_mask_list_node(board_etree, scenario_etree, allocation_etree, mask_list)
  File "/home/mleon/Desktop/development/acrn/acrn-hypervisor/misc/config_tools/static_allocators/clos.py", line 125, in creat_mask_list_node
    allocation_hv_node = common.append_node("//hv", None, allocation_etree, id = vm_id)
NameError: name 'vm_id' is not defined
make[1]: *** No rule to make target '/home/mleon/Desktop/development/acrn/acrn-hypervisor/build/hypervisor/configs/allocation.xml', needed by '/home/mleon/Desktop/development/acrn/acrn-hypervisor/build/hypervisor/configs/unified.xml'.  Stop.
make[1]: Leaving directory '/home/mleon/Desktop/development/acrn/acrn-hypervisor/hypervisor'
make: *** [Makefile:133: hvdefconfig] Error 2

I get the same error when I use my custom board config and custom board scenario. 

When I tried building v2.7 with both of the configs attached to this issue, I get:

Error: board config: The board xml file does not match scenario xml file.

When I tried building v2.7 with both of my custom configs, it succeeds.

I am not sure what I am doing wrong here.

leonmatt, 2022-04-26:
@gvancuts After doing a clean clone of the repo, using my-NUC11PAHi5 and shared.xml config files, both of which are attached to this issue, I get the following error:

Board XML is fetched from /home/mleon/Desktop/development/configs_modified/temporary/my-NUC11PAHi5.txt
Scenario XML is being fetched from /home/mleon/Desktop/development/configs_modified/temporary/shared.txt
Traceback (most recent call last):
File "/home/mleon/Desktop/development/acrn/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 41, in
main(args)
File "/home/mleon/Desktop/development/acrn/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 31, in main
module.fn(board_etree, scenario_etree, allocation_etree)
File "/home/mleon/Desktop/development/acrn/acrn-hypervisor/misc/config_tools/static_allocators/clos.py", line 149, in fn
creat_mask_list_node(board_etree, scenario_etree, allocation_etree, mask_list)
File "/home/mleon/Desktop/development/acrn/acrn-hypervisor/misc/config_tools/static_allocators/clos.py", line 125, in creat_mask_list_node
allocation_hv_node = common.append_node("//hv", None, allocation_etree, id = vm_id)
NameError: name 'vm_id' is not defined
make[1]: *** No rule to make target '/home/mleon/Desktop/development/acrn/acrn-hypervisor/build/hypervisor/configs/allocation.xml', needed by '/home/mleon/Desktop/development/acrn/acrn-hypervisor/build/hypervisor/configs/unified.xml'. Stop.
make[1]: Leaving directory '/home/mleon/Desktop/development/acrn/acrn-hypervisor/hypervisor'
make: *** [Makefile:133: hvdefconfig] Error 2

I get the same error when I use my custom board config and custom board scenario.

When I tried building v2.7 with both of the configs attached to this issue, I get:

Error: board config: The board xml file does not match scenario xml file.

When I tried building v2.7 with both of my custom configs, it succeeds.

gvancuts, 2022-04-27:
> @gvancuts After doing a clean clone of the repo, using my-NUC11PAHi5 and shared.xml config files, both of which are attached to this issue, I get the following error:
> 

We have been making heavy changes to the `board_inspector` and `configurator` tools so make sure that the board and scenario files you try to compile were created with the tools from the same code version. Could it be that you tried to re-use your 2.7 XML files with the latest code?

leonmatt, 2022-04-27:
I tried the XML files attached to this issue with a clean clone of the repo. I figured those would work because they successfully compiled for everyone else here. I will try installing the latest board_inspector and generate new configs and retry compiling.

NanlinXie, 2022-09-07:
@gvancuts , could we close this issue?

gvancuts, 2022-09-07:
Not sure, let's close it for now. We can re-open later if needed.

closed 2022-09-07 by gvancuts