projectacrn/acrn-hypervisor #7082: I had forgotten to reboot the platform before generating the board XML, here is the latest file (and `board_inspector` output):

gvancuts, 2022-01-31:
Errors while generating the board XML, `board_inspector` output:
```
acrn@acrn-tgl-nuc:~$ cat /proc/cmdline 
BOOT_IMAGE=/boot/vmlinuz-5.13.0-27-generic root=UUID=50131b44-fc2b-40c0-9782-3067f6e42c61 ro quiet splash idle=nomwait iomem=relaxed intel_idle.max_cstate=0 intel_pstate=disable vt.handoff=7
acrn@acrn-tgl-nuc:~$ sudo board_inspector.py  my-NUC11PAHi5
[sudo] password for acrn: 
WARNING:root:Parse ACPI DSDT/SSDT failed: [Errno 1] Operation not permitted: '/dev/mem'
WARNING:root:Will not extract information from ACPI DSDT/SSDT
my-NUC11PAHi5.xml saved successfully!
```
[my-NUC11PAHi5.txt](https://github.com/projectacrn/acrn-hypervisor/files/7950565/my-NUC11PAHi5.txt)

_Originally posted by @gvancuts in https://github.com/projectacrn/acrn-hypervisor/issues/7069#issuecomment-1023181833_

gvancuts, 2022-01-31:
Additional information:
```
acrn@acrn-vecow:~$ sudo mokutil --sb-state 
SecureBoot disabled
Platform is in Setup Mode
acrn@acrn-vecow:~$ sudo board_inspector.py vecow-no-secureboot
CPU turbo is not enabled!
RTCT table has been saved to RTCT successfully!
vecow-no-secureboot.xml saved successfully!
```

Not sure whether SecureBoot **must** be disabled when running the `board_inspector` tool, but if so, perhaps we can add a check in it to warn the user when it's enabled? The only way I found we can determine this from a running system is using `sudo mokutil --sb-state`.

fuzhongl, 2022-07-05:
Highlight in following link: Disable Secure Boot
https://projectacrn.github.io/latest/getting-started/getting-started.html#configure-target-bios-settings
`Disable Secure Boot. This setting simplifies the steps for this example.`

closed 2022-07-05 by fuzhongl