projectacrn/acrn-hypervisor #7087: Do we or plan to support virtio mmio bus?

archersmind, 2022-02-07:
It seems that `acrn` only supports `PCI` bus for `virtio`, my question is do we have any plan to support `mmio` bus in the future? Any roadmap planned?

Many thanks to the acrn team.

gvancuts, 2022-02-07:
@anthonyzxu @terryzouhao can you take a look at this question please?

ywan170, 2022-02-08:
Hi Alan,

We have no plan to support mmio based virtio devices as we only support X86 so far.
Once the ACRN supports the ARM platform then this is a must-have feature.
BTW, may I know what is your usage for this requirement?

Thanks
Yu

archersmind, 2022-02-08:
Thank you Yu for the reply,

Actually we're conducting some research to bring up acrn on top of ARM platform, so the the question would be ARM support is in your roadmap yet?

terryzouhao, 2022-02-08:
@archersmind, Glad to see you are interested and researching enabling ACRN on ARM platform. Basically, ACRN is designed to support multi-architectures, including ARM, but only enabled and verified on IA platform yet. As open source project, ACRN is open and welcome everybody in community to contribute ARM enabling, we can discuss more for any idea/proposal in mailing list for detailed technical discussing first, then come to TSC for design/arch approval next.

archersmind, 2022-02-08:
Thank you guys, totally understood!

closed 2022-02-08 by archersmind