projectacrn/acrn-hypervisor #7150: Inconsistent use of \n and \r\n in message output

dbkinder, 2022-02-25:
I noticed some of the messages written when running the User VM launch script show up only with a linefeed and no return so they aren't left-justified. That seems rather odd for a Linux system.

The result is console logging output that's not consistently left-justified.  Here's a portion of the output on the console
```
pci init virtio-blk
pci init virtio-net
tpm: init_vtpm2:Invalid socket path!
                                    basl_fwrite_madt, guest_cpu_bitmask: 0x3
                                                                            basl_fwrite_madt, vcpu_id=0x0, pcpu_id=0x0, lapic_id=0x0
                                                                                                                                    basl_fwrite_madt, vcpu_id=0x1, pcpu_id=0x1, lapic_id=0x2
                                                                                                                                                                                            acrn_sw_load
SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 0x200000 copied to addr 0x7fa3bfe00000
                                                                                                 SW_LOAD: build e820 4 entries to addr: 0x7fa2c00ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
```

For example, this call to pr_notice has a ``\r\n`` in its output string:  
```
devicemodel/hw/pci/core.c  pr_notice("pci init %s\r\n", fi->fi_name);
```
while this call only has a ``\n``:
```
devicemodel/hw/platform/tpm/tpm.c:              WPRINTF("Invalid socket path!\n");
```

This call has a ``\r\n`` in it:
```
devicemodel/core/sw_load_common.c:      pr_info("SW_LOAD: build e820 %d entries to addr: %p\r\n",
```
while this call only has a ``\n``:
```
devicemodel/hw/platform/acpi/acpi.c:            pr_info("%s, vcpu_id=0x%x, pcpu_id=0x%x, lapic_id=0x%x\n", __func__, i, pcpu_id, lapic_id);
```
Do all these pr_xxxx and WPRINTF calls need to use \r\n  to display properly, or is some other configuration causing this behavior?

gvancuts, 2022-02-25:
This may be better followed-up on in a separate issue but why are we using a mix of `WPRINTF` and `pr_err`, `pr_info`, etc. functions?

dbkinder, 2022-02-25:
> This may be better followed-up on in a separate issue but why are we using a mix of `WPRINTF` and `pr_err`, `pr_info`, etc. functions?

I'm not sure why, bit It looks like WPRINTF is defined (multiple times) as a macro calling pr_err. :)
```
#define WPRINTF(params) (pr_err params)
```

dbkinder, 2022-03-21:
Another example of this missing linefeed behavior happens if you use ``poweroff`` to shutdown the User VM:

```
ubuntu@ubuntu:~$ sudo poweroff
ubuntu@ubuntu:~$ Unhandled ps2 keyboard command 0xf6
                                                    vm_suspend: setting VM state to POWEROFF
            VM state changed from[ RUNNING ] to [ POWEROFF ]
                                                            VM loop exit
                                                                        pci deinit hostbridge
             pci deinit lpc
                           pci deinit virtio-console
                                                    pci deinit virtio-blk
pci deinit virtio-net
vtnet tx thread closing...
mngr_client_new: Failed to accept from fd 37, err: Invalid argument
Stop listening 37...
Stop polling 37...
No correct pm notify channel given
acrn@vecow:~/acrn-work$ 
```

fuzhongl, 2022-07-05:
PR merged into master branch:
https://github.com/projectacrn/acrn-hypervisor/pull/7861

dbkinder, 2022-07-28:
This issue is not resolved.  I'm still seeing wonky output with missing returns when launching the user VM via the launch script created by the ACRN configurator.

fuzhongl, 2022-07-28:
> @fuzhongl @NanlinXie This issue is not resolved. I'm still seeing wonky output with missing returns when I follow the GSG and get to launching the user VM via the launch script created by the ACRN configurator, as initially reported.

@dbkinder The PR only merged into master branch; you are using Master branch, not  v3.0 branch, right?

dbkinder, 2022-07-29:
> @dbkinder The PR only merged into master branch; you are using Master branch, not v3.0 branch, right?

I'm using the v3.0 release to walk through the GSG.  Sigh.  I'll close this PR and check again later.

closed 2022-07-29 by dbkinder