projectacrn/acrn-hypervisor #7156: Predefined tgl-vecow-spc-7100-Corei7 scenario and launch XML disagree

dbkinder, 2022-02-27:
The predefined XML files found in ``/misc/config_tools/data/tgl-vecow-spc-7100-Corei7`` disagree about the system configuration.
The ``shared.xml`` defines  a SERVICE_VM (ID 0) where the root for booting is ``root=/dev/sda3`` and the image name is ``Linux_bzImage``, and defines one POST_LAUNCHED_VM (ID 1).  

The ``shared_launch_2user_vm.xml`` defines VM ID 1 as a WINDOWS VM where the image can be found at ``./win10-ltsc.img``
and also defines a HARD_RT VM with VM ID 2 with it's image at
``./core-image-weston-intel-corei7-64.wic``

VM ID 2 is not defined in the scenario.xml so the launch xml can't define more VMs than what's in the scenario.xml, right?

There's also no documentation about these configuration files describing what they to do and that image files are needed for ``Linux_bzImage``, ``./win10-ltsc.img``, and ``./core-image-weston-intel-corei7-64.wic``.  We should document these predefined xml files and how to work with them.

dbkinder, 2022-02-28:
@junjiemao1 should the configurator catch a launch XML with more VMs than defined in the scenario XML?

junjiemao1, 2022-02-28:
Having duplicated information in scenario and launch XMLs leads to potential inconsistency here and there, and this disagreement of the number of VMs is one of them.

In v3.0 we'll get rid of these by merging data in launch XMLs into the scenario XML.

closed 2022-07-05 by fuzhongl