projectacrn/acrn-hypervisor #7161: config_tool: ./acrn_configurator.py fails with ImportError

dbkinder, 2022-02-28:
On the master branch, running ``./acrn_configurator.py`` fails with this error:
```
dbkinder@dbkinder-ThinkPad-X220:~/acrn-work/acrn-hypervisor/misc/config_tools/config_app$ ./acrn_configurator.py
Traceback (most recent call last):
  File "./acrn_configurator.py", line 34, in <module>
    from views import CONFIG_APP
  File "/home/dbkinder/acrn-work/acrn-hypervisor/misc/config_tools/config_app/views.py", line 26, in <module>
    from launch_config.launch_cfg_gen import get_launch_item_values
ImportError: cannot import name 'get_launch_item_values' from 'launch_config.launch_cfg_gen' (/home/dbkinder/acrn-work/acrn-hypervisor/misc/config_tools/config_app/../launch_config/launch_cfg_gen.py)
dbkinder@dbkinder-ThinkPad-X220:~/acrn-work/acrn-hypervisor/misc/config_tools/config_app$
```

dbkinder, 2022-02-28:
@junjiemao1 does acrn_configurator.py launch for you without error?

gvancuts, 2022-02-28:
I see the same problem. I did a `git bisect` and this is the commit that introduced this problem: https://github.com/projectacrn/acrn-hypervisor/commit/2450b4c77f2f59d267c31e62b3287c6f0d5d2876

junjiemao1, 2022-03-01:
This configurator is supposed to be replaced with its next-generation in a week or two, so I did not consider the compatibility between the rewritten launch script generator and the old configurator.

If necessary, we can revert back to the old one till the new configurator is merged.

dbkinder, 2022-03-02:
> If necessary, we can revert back to the old one till the new configurator is merged.

If the old configurator will run and create correct XML and launch script output, I'd recommend putting it back so the GSG instructions will work on the master branch. If it won't, then sounds like we'll need to wait for the new configurator.

gvancuts, 2022-03-02:
Let's live with that for now then. Do you know when things will be working again @junjiemao1 ?