projectacrn/acrn-hypervisor #7168: Copying kernel file in GSG for Partitioned Mode

xsltzx, 2022-03-01:
Hello，I have some doubts about this step in GSG for partitioned mode

[image]

The kernel release info means the original kernel of the target computer itself(for example 5.4.0-89-generic), or it means the Service VM kernel(for example vmlinuz-5.10.65-acrn-service-vm) as well as initrd.img's version. I want to know which version should I choose for partitioned mode, 

I am using version 2.7.

thanks.

gvancuts, 2022-03-01:
It's the original kernel you want to use in the pre-launched VM, not the Service VM one.

xsltzx, 2022-03-01:
> It's the original kernel you want to use in the pre-launched VM, not the Service VM one.

Oh, thanks for your reply and now I know that.

closed 2022-03-08 by fuzhongl