projectacrn/acrn-hypervisor #7210: Virtio-gpu adapter support in ACRN

psun3x, 2022-03-22:
Implement virtio-gpu basic 2D display in fixed resolution with hardware cursor support. The framebuffer from guest vm can be shown in a window of service vm GUI system(X11 or Wayland). Virtio-gpu device provide to 2 interface for displaying operation, VGA interface and virtio-gpu command interface.

psun3x, 2022-03-24:
[External_System_ID] ACRN-7626

NanlinXie, 2022-03-31:
Doc Impact Analysis: 

- [x] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [x] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [x] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

closed 2022-07-05 by fuzhongl